OAID/Tengine #382: 模型输出结果不一致

mengfu188, 2020-08-25:
看到了 #371 ，想添加Mobilefacenets，但是模型转换之后，输入全0，输出结果不一致

仓库：https://github.com/deepinsight/insightface/
模型：https://github.com/deepinsight/insightface/wiki/Model-Zoo#34-mobilefacenetarcfacems1m-refine-v1
python代码：
```python
from __future__ import absolute_import
from __future__ import division
from __future__ import print_function

import numpy as np
import mxnet as mx

def do_flip(data):
  for idx in range(data.shape[0]):
    data[idx,:,:] = np.fliplr(data[idx,:,:])

def get_model(ctx, image_size, model_str, layer):
  _vec = model_str.split(',')
  assert len(_vec)==2
  prefix = _vec[0]
  epoch = int(_vec[1])
  print('loading',prefix, epoch)
  sym, arg_params, aux_params = mx.model.load_checkpoint(prefix, epoch)
  all_layers = sym.get_internals()
  sym = all_layers[layer+'_output']

  model = mx.mod.Module(symbol=sym, context=ctx, label_names = None)
  model.bind(data_shapes=[('data', (1, 3, image_size[0], image_size[1]))])
  model.set_params(arg_params, aux_params)

  return model

model = get_model(mx.cpu(), (112,112), 'F:\\cmf\\models\\model-y1-test2\\model,0', 'fc1')
img = np.zeros(shape=(3,112,112), dtype=np.uint)
input_blob = np.expand_dims(img, axis=0)
data = mx.nd.array(input_blob)
db = mx.io.DataBatch(data=(data,))
model.forward(db, is_train=False)
embedding = model.get_outputs()[0].asnumpy()
print(embedding)

```
结果：
```
[[ 0.02154729 -0.4162834  -0.09101282  0.7223283  -0.11585438  0.0611931
  -0.15061818 -0.16417424  0.03914393 -0.22145584  0.03533301  0.13380608
  -0.24567278 -0.35991496 -0.17583792 -0.14100203  0.17048325  0.2689074
  -0.32794398 -0.11333947 -0.02633217 -0.47033542 -0.10491757  0.13647714
  -0.14120585  0.2601153  -0.2262695   0.15353546 -0.24051094  0.19708422
   0.11372234 -0.106502   -0.04841888  0.24713843  0.18096168  0.17739116
  -0.26957372  0.52657896  0.51086056 -0.34206912 -0.5117914   0.45784622
   0.31155518  0.01418302 -0.2696568  -0.14567082 -0.03723009  0.39327598
  -0.45988983 -0.1328338  -0.01042032 -0.7263899  -0.60506123  0.04480996
   0.00260787 -0.22903123  0.06622976  0.06381423 -0.07263767 -0.02013894
  -0.05507097  0.05374932  0.4380721  -0.264139   -0.23919213  0.41576117
  -0.05746923 -0.22019155 -0.1482154  -0.04492995  0.6597815  -0.14962853
  -0.18185823 -0.24876197  0.05852314  0.36671737  0.12722713 -0.17162843
   0.01947528 -0.3874434  -0.42982838  0.21893841  0.07214668 -0.18667988
  -0.07072115 -0.26045492  0.14012316  0.21946907 -0.28807697  0.12465948
   0.0042928   0.07700975  0.15083246  0.13297793  0.11155006  0.4361202
   0.19571272 -0.48228177 -0.71047276 -0.5125456  -0.4758979  -0.2889716
  -0.01963822  0.48873454  0.2459967   0.18166065  0.30512288  0.14568155
   0.20536348 -0.23129442  0.10100278  0.22046189  0.42669702 -0.03751966
   0.61379117  0.34831098  0.51180166 -0.22458424  0.1418204  -0.22925594
  -0.53545964 -0.12080469 -0.2029418   0.12714365 -0.26134396  0.08022993
  -0.33410394  0.42091098]]
```
使用的转换工具：Tengine官网提供的工具
cpp代码
[... 344 lines cut ...]
2.787490, 109
-0.749577, 110
0.710049, 111
0.157990, 112
-1.053664, 113
4.362529, 114
-0.256790, 115
-0.434527, 116
1.626453, 117
-1.711483, 118
-1.843128, 119
-0.476325, 120
2.609643, 121
-0.704538, 122
-5.628683, 123
1.847601, 124
-3.102808, 125
-3.541050, 126
-1.755906, 127
output_size 128
--------------------------------------

Process finished with exit code 0

```

BUG1989, 2020-08-25:
请问，与原始框架相比，哪个结果更接近正确呢？

mengfu188, 2020-08-25:
哪个结果？？是哪一层结果吗？

mengfu188, 2020-08-25:
现在出来的效果是不同的人，结果都很相似

mengfu188, 2020-08-25:
我写了个和Tengine debug类似的脚本，可以输出mxnet每一层的结果 https://blog.csdn.net/night_mfc/article/details/108219908 希望对你有所帮助

BUG1989, 2020-08-25:
谢谢您的分享，我会安排工程师，尽快解决这个问题

wei8171023, 2022-10-18:
我这用的是相同的mobilefacenet-mxnet模型，转到onnx后再转tengine，输入都是0，输出结果是一致的。