OAID/Tengine #396: linux x86_64运行 example中的 mobilenet_ssd结果错误

wuyuanmm, 2020-09-04:
模型从tengine的zoo中下载，图片使用截图
以下是编译的情况：

-- CMAKE_INSTALL_PREFIX = /home/wuyuan/Tengine-tengine-lite/build/install
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- TENGINE: OpenMP was found.
-- TENGINE: OpenMP support enabled.
-- TENGINE: 'OpenMP_C_FOUND' is TRUE.
-- TENGINE: 'TENGINE_LINKING_LIBS' is OpenMP::OpenMP_C.
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") 
-- 
-- 
-- Infomation Summary:
-- 
-- CMake infomation:
--   - CMake version:              3.11.3
--   - CMake generator:            Unix Makefiles
--   - CMake building tools:       /usr/bin/make
--   - Target System:              Linux
--   - Target CPU arch:            X86
--   - Target CPU bus width:       64 Bit
-- 
-- tengine-lite toolchain infomation:
--   Cross compiling: FALSE
--   C/C++ compilier:
--     - C   standard version:     C99
--     - C   standard required:    TRUE
--     - C   standard extensions:  ON
--     - C   compilier version:    7.5.0
--     - C   compilier:            /usr/bin/cc
--     - C++ standard version:     C++11
--     - C++ standard required:    TRUE
--     - C++ standard extensions:  OFF
--     - C++ compilier version:    7.5.0
--     - C++ compilier:            /usr/bin/c++
--   C/C++ compilier flags:
--     - C   compilier flags:      
--     - C++ compilier flags:      
--   OpenMP:
--     - OpenMP was found:         YES
--     - OpenMP version:           4.5
-- 
-- tengine-lite building infomation:
--   - Project source path is:     /home/wuyuan/Tengine-tengine-lite
--   - Project building path is:   /home/wuyuan/Tengine-tengine-lite/build
-- 
-- tengine-lite other infomation:
--   Package install path:         /home/wuyuan/Tengine-tengine-lite/build/install
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wuyuan/Tengine-tengine-lite/build

模型和测试图片以及结果在[https://pan.baidu.com/s/19H3U2hFgl_X5c2y-mdFlfA](url)

wuyuanmm, 2020-09-04:
提取码， ssss

BUG1989, 2020-09-04:
@wuyuanmm 您好，请问原始模型是 Caffe 的，还是 TensorFlow 的呢？

wuyuanmm, 2020-09-04:
您好，感谢您的回复，原始模型 是从 tengine的模型仓库下载的，应该事caffe的

BUG1989, 2020-09-04:
@wuyuanmm 您好，我修复了之前的softmax 的bug，请更新下代码，如果还是异常，请试试这里的模型。谢谢
[参考模型](https://pan.baidu.com/s/1NH5sE4hPkkaQlzarFZvKNA)(提取码：4far)

closed 2020-09-05 by BUG1989