OAID/Tengine #413: YOLOV4 problems /yolov4-tiny require

w840401, 2020-09-25:
目前使用Tengine_Convert_Tool 成功將darknet的yolov3/yolov4轉換成tm_file模型檔案
在使用yolov3-tiny時成功運行，不過在使用yolov4時，出現以下圖片錯誤:
請求該如何解決?
另外想請求yolov4-tiny以及yolov4-tiny-int8的版本?未來有機會實現嗎?
[image]

下圖為成功運行yolov3-tiny版本:
[image]

w840401, 2020-09-25:
發現使用tm_yolact 配上yolov3.tm.file 也是顯示 "已砍掉"
使用examples中的tm_yolact 配上yolov3-tiny.tmfile 顯示 "程式記憶體區段錯誤(核心已傾印)"
只有使用tm_yolov3_tiny配上yolov3-tiny.tmfile 有輸出結果，不過測試另外一張圖時(Bus)那張，也框不出BUS，如下:
想請教這是BUG嗎 有有誰能幫幫我?
[image]

zjd1988, 2020-09-25:
@w840401 你好，我晚上试试，你方便传下你的模型和测试图片文件吗？

w840401, 2020-09-25:
> @ w840401你好，我晚上試試，你方便傳下你的模型和測試圖片文件嗎？

感謝回覆
我的模型是使用您們 https://github.com/OAID/Tengine-Convert-Tools  轉換工具轉成的tm_file
至於模型就是darknet上 使用COCO 80類所訓練出來的 檔案有點大 傳不上來...
[這是darknet上的官方.cfg](https://github.com/AlexeyAB/darknet/tree/master/cfg)
[這是yolov4.weights](https://github.com/AlexeyAB/darknet/releases/download/darknet_yolo_v3_optimal/yolov4.weights)
yolo3的忘記網址在哪了..不過我希望使用yolo4 以及**yolov4-tiny / yolov4-tiny-int8**
感謝大佬!

zjd1988, 2020-09-25:
@w840401  我本地测试了下u版的yolov4模型，是能正常运行的
[image]
[image]

你要不把tmfile模型上传到百度云盘，我下载试试。

w840401, 2020-09-25:
> @w840401 我本地测试了下u版的yolov4模型，是能正常运行的
> [image]
> [image]
> 
> 你要不把tmfile模型上传到百度云盘，我下载试试。

u版的意思是啥意思!?
我這邊沒法登入百度....
我能+你QQ 傳給你嗎? 不好意思

zjd1988, 2020-09-25:
@w840401 我的qq942002795

w840401, 2020-09-26:
經過重開機後，yolov3 / yolov3-tiny /yolov4 皆可正常執行
模型都是採用darknet官方coco訓練的
也都是使用自己編譯後的convert_tool 轉換的
不過yolov4-tiny照理來說應該能實踐於tm_yolov4...
剩下求救於yolov4-tiny/yolov4-tiny-int8 的實踐   求指教 感恩.

zjd1988, 2020-09-27:
@w840401 @BUG1989 darknet中的route算子在yolov4-tiny中有新增属性，目前的convert tools暂时不支持，导致在解析权重weights文件时出错 https://github.com/ceccocats/tkDNN/issues/59

closed 2020-10-14 by w840401