OAID/Tengine #440: 赏金任务，第二期（快来薅羊毛~~）

BUG1989, 2020-10-18:
## **任务激励**

- 达成任意一项任务即成为 Tengine 开源贡献者、活动贡献者认证证书；
- 根据达成任务的难度还有额外小礼品赠送
  - 简单：T桖 + 多功能充电线 
  - 一般：T桖 + 多功能充电线 + 帽子
  - 困难：T桖 + 多功能充电线 + 机械键盘

## **任务详情**

### 示例类（C/C++）

- [ ] 基于 Tengine  Lite 最新版本，实现 alphapose demo；（难度：困难）

### API类

- [ ] 基于现有 C++ API 重构当前 Python API，提升易用性，并更新相关 demo。（难度：困难）

### 算子类（共3个）

- [ ] 待定，欢迎大家补充。（难度：一般）
- [ ] 待定，欢迎大家补充。（难度：一般）
- [ ] 待定，欢迎大家补充。（难度：一般）

### 优化类

- [ ] 基于 Tengine Lite 最新版本，实现 armv7/v8 conv im2col+sgemm int8 性能算子（intrinsic 即可）。（难度：困难）
- [ ] 基于 Tengine Lite 最新版本，实现 armv7/v8 convdw3x3s1/s2 int8 性能算子（intrinsic 即可）。（难度：一般）
- [ ] 基于 Tengine Lite 最新版本，实现 armv7/v8 fc int8 性能算子（intrinsic 即可）。（难度：一般）
- [ ] 基于 Tengine Lite 最新版本，实现 x86 avx2 im2col+sgemm int8 性能算子 （intrinsic 即可）。（难度：困难）

[参考模型 mobilenet_v1_int8.tmfile ](https://pan.baidu.com/s/1NH5sE4hPkkaQlzarFZvKNA)(提取码：4far)

### 编译类

- [ ] 基于 Tengine Lite 最新版本，实现在 华为鸿蒙OS 编译生成 libtengine-lite.so，并执行任意 demo 即可。（难度：一般）

### 文档类

- [ ] 基于 Tengine Lite 最新版本，在 examples/README.md 中补充 OCR 示例运行方式及结果。（难度：简单）
- [ ] 基于 Tengine Lite 最新版本，在 examples/README.md 中补充 yolov4_tiny 示例运行方式及结果。（难度：简单）
- [ ] 基于 Tengine Lite 最新版本，在 所有的 README.md 中修复 ≥ 3处文本错误。（难度：简单）
- [ ] 基于 Tengine Lite 最新版本，在 所有的 README.md 中修复 ≥ 3处文本错误。（难度：简单）
- [ ] 基于 Tengine Lite 最新版本，在 所有的 README.md 中修复 ≥ 3处文本错误。（难度：简单）

[参考模型](https://pan.baidu.com/s/1NH5sE4hPkkaQlzarFZvKNA)(提取码：4far)

------

**第二期悬赏任务说明：**

1. 所有悬赏任务均以 Pull Request 形式提交，无需提交 tmfile 模型文件、图片（若需展示，请使用外部链接）；
2. 奖励标准：提交时间和任务完成质量，评选规则由 Tengine 开源委员会评选和决定；
3. 首期悬赏任务截止时间：2020.12.31；奖励和贡献者证书在 2020.12.31 统一发送；
4. 奖励发送&加入贡献者交流群，请添加Tengine小助手微信号：Tengine666 备注：任务
5.  Tengine 开源委员会保留最终解释权。

zjd1988, 2020-10-19:
@BUG1989  大佬，yolov4_tiny示例需要转换得到对应的模型，这个需要合入https://github.com/OAID/Tengine-Convert-Tools/pull/18

BUG1989, 2020-10-19:
@zjd1988 已合并，您能提供下 yolov4_tiny 的 tmfile 文件吗？谢谢.

zjd1988, 2020-10-19:
@BUG1989  嗯嗯，好的，我今天抽空把readme 补充下。附件模型文件
[yolov4-tiny.tmfile.zip](https://github.com/OAID/Tengine/files/5398643/yolov4-tiny.tmfile.zip)

BUG1989, 2020-10-19:
任务完成：更新 benchmark/README.md，新增 树莓派3B 数据（64位版本，Cortex-A53）单线程数据
#444

BUG1989, 2020-10-20:
任务完成：更新 benchmark/README.md，新增 树莓派4 数据（64位版本，Cortex-A53）单线程数据
#447

BUG1989, 2020-10-20:
任务完成：基于 Tengine Lite 最新版本，在 examples/README.md 中补充 yolov4_tiny 示例运行方式及结果
#445

BUG1989, 2020-10-20:
任务完成：基于 Tengine Lite 最新版本，在 所有的 README.md 中修复 ≥ 3处文本错误
#446

Bridgette2020, 2020-10-20:
@zgd1988 @linglan2020 @xingxingRealzyx，请添加Tengine小助手微信号：Tengine666 备注：任务，活动结束后好及时与您联系，发送奖品。已经添加请私聊一下，方便工作人员记录完成信息。

LEMON-LM, 2020-10-20:
@zjd1988 @linglan2020 @xingxingRealzyx 请添加Tengine小助手微信号：Tengine666 备注：任务，活动结束后好及时与您联系，发送奖品。已经添加请私聊一下，方便工作人员记录完成信息

BUG1989, 2020-10-20:
任务完成：基于 Tengine Lite 最新版本，在 examples/README.md 中补充 OCR 示例运行方式及结果
#450

BUG1989, 2020-10-21:
任务完成：基于 Tengine Lite 最新版本，在 examples/README.md 中补充 OCR 示例运行方式及结果
#451

BUG1989, 2020-10-21:
任务完成：基于 Tengine Lite 最新版本，在 所有的 README.md 中修复 ≥ 3处文本错误
#449

BUG1989, 2020-10-25:
任务完成：基于 Tengine Lite 最新版本，在 所有的 README.md 中修复 ≥ 3处文本错误
#454

BUG1989, 2020-10-27:
任务完成：基于 Tengine Lite 最新版本，实现 alphapose demo
#453

zchrissirhcz, 2020-10-30:
@BUG1989 

>基于 Tengine Lite 最新版本，实现在 华为鸿蒙OS 编译生成 libtengine-lite.so，并执行任意 demo 即可。（难度：一般）

提交了PR： https://github.com/OAID/Tengine/pull/461

BUG1989, 2020-10-30:
任务完成：基于 Tengine Lite 最新版本，实现在 华为鸿蒙OS 编译生成 libtengine-lite.so，并执行任意 demo 即可。
#461 @zchrissirhcz

BUG1989, 2020-11-04:
任务完成：基于 Tengine Lite 最新版本，实现 armv7/v8 fc int8 性能算子（intrinsic 即可）。
#467

BUG1989, 2020-12-01:
任务完成：基于 Tengine Lite 最新版本，实现 x86 avx2 im2col+sgemm int8 性能算子 （intrinsic 即可）。
#490

BUG1989, 2020-12-01:
任务完成：基于 Tengine Lite 最新版本，实现 armv7/v8 convdw3x3s1/s2 int8 性能算子（intrinsic 即可）。
#470

BUG1989, 2020-12-01:
任务完成：基于 Tengine Lite 最新版本，实现 armv7/v8 convdw3x3s1/s2 int8 性能算子（intrinsic 即可）。
#478