OAID/Tengine #452: Failed to build with HarmonyOS Toolchain

zchrissirhcz, 2020-10-22:
尝试编译Tengine-lite的鸿蒙系统上的库。

已经安装了DevEco Studio，下载了Native工具链，基于cmake编译hello world已经走通。

编译Tengine，用的编译脚本`build/ohos-arm64-v8a.bat`内容如下：
```batch
@echo off

set OHOS_NDK=E:/soft/Huawei/sdk/native/3.0.0.80
set TOOLCHAIN=%OHOS_NDK%/build/cmake/ohos.toolchain.cmake

REM echo "=== TOOLCHAIN is: $TOOLCHAIN"

set BUILD_DIR=ohos-arm64-v8a
if not exist %BUILD_DIR% md %BUILD_DIR%
cd %BUILD_DIR%

cmake -G Ninja ^
    -DCMAKE_TOOLCHAIN_FILE=%TOOLCHAIN% ^
    -DOHOS_ARCH="arm64-v8a" ^
    -DCMAKE_BUILD_TYPE=Release ^
    ../..

ninja

cd ..
```

目前遇到报错是，汇编相关的代码，和-fPIC编译选项有冲突：

[image]

```
[322/406] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0.S.obj
FAILED: src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0.S.obj
E:\soft\Huawei\sdk\native\3.0.0.80\llvm\bin\clang.exe --sysroot=E:/soft/Huawei/sdk/native/3.0.0.80/sysroot -DCONFIG_TENGINE_SERIALIZER -Dtengine_lite_EXPORTS -I../../include -I../../src/op -I../../src/serializer/tm -x assembler-with-cpp -O3 -DNDEBUG -fPIC   -g -O0 -MD -MT src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0.S.obj -MF src\CMakeFiles\tengine-lite.dir\dev\cpu\op\aarch64\dw_k3s2p0.S.obj.d -o src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0.S.obj -c ../../src/dev/cpu/op/aarch64/dw_k3s2p0.S
clang: error: unsupported option '-fPIC' for target 'x86_64-w64-windows-gnu'
clang: error: unsupported option '-fPIC' for target 'x86_64-w64-windows-gnu'
[325/406] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0p1.S.obj
FAILED: src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0p1.S.obj
E:\soft\Huawei\sdk\native\3.0.0.80\llvm\bin\clang.exe --sysroot=E:/soft/Huawei/sdk/native/3.0.0.80/sysroot -DCONFIG_TENGINE_SERIALIZER -Dtengine_lite_EXPORTS -I../../include -I../../src/op -I../../src/serializer/tm -x assembler-with-cpp -O3 -DNDEBUG -fPIC   -g -O0 -MD -MT src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0p1.S.obj -MF src\CMakeFiles\tengine-lite.dir\dev\cpu\op\aarch64\dw_k3s2p0p1.S.obj.d -o src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0p1.S.obj -c ../../src/dev/cpu/op/aarch64/dw_k3s2p0p1.S
clang: error: unsupported option '-fPIC' for target 'x86_64-w64-windows-gnu'
clang: error: unsupported option '-fPIC' for target 'x86_64-w64-windows-gnu'
[331/406] Building C object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/tanh/cortex-a/tanh_kernel_arm.c.obj
ninja: build stopped: subcommand failed.
```

zchrissirhcz, 2020-10-22:
@kalcohol 大佬有空可以帮忙看看吗~

kalcohol, 2020-10-22:
> @kalcohol 大佬有空可以帮忙看看吗~

哪里可以下载工具链？我也下载试试编一下

zchrissirhcz, 2020-10-22:
我写了一个notes，可以参考
https://github.com/zchrissirhcz/cmake_examples/blob/master/ohos_ndk_example/detail.md

cccxinli, 2020-10-30:
clang --target=aarch64-linux-ohos  可以解决 clang: error: unsupported option '-fPIC' for target 'x86_64-w64-windows-gnu' 的问题。@zchrissirhcz

zchrissirhcz, 2020-10-30:
@cccxinli Thanks! I'll have a try

zchrissirhcz, 2020-10-30:
Will make a PR.

closed 2020-10-30 by zchrissirhcz