OAID/Tengine #489: 赏金任务，第三期（快来薅羊毛~~）

lyuchuny3, 2020-11-30:
## 任务激励
- 达成任意一项任务即成为 Tengine 开源贡献者、活动贡献者认证证书；
- 根据达成任务的难度还有额外小礼品赠送
  - 简单：T恤 + 多功能充电线 
  - 一般：T恤 + 多功能充电线 + 帽子
  - 困难：T恤 + 多功能充电线 + 机械键盘

## 任务详情
### **算子类（C++- float32）**
此类算子添加参考文档 [如何快速添加算子](https://github.com/OAID/AutoKernel/blob/main/doc/how_to_add_op.md), 测试脚本放在 `autokernel_plugin/tests`目录下
- [ ] 基于最新版本，实现 avgpooling的移植，验证正确性（难度：简单）
- [ ] 基于最新版本，实现 maxpooling的移植，验证正确性（难度：简单）
- [ ] 基于最新版本，实现 fc算子的移植，验证正确性（难度：简单）
- [ ] 基于最新版本，实现 dw_conv(depthwise_convolution)的移植，验证正确性（难度：简单）
- [ ] 其他算子待定，欢迎大家待补充（难度：简单）
  
### **后端类（C/C++）**
文档提交在 `doc`目录下
- [ ] 实现任意算子的其他后端实现(比如cuda)，并提供文档说明（doc/xxx_backend.md)（难度：一般）

### **优化类（C/C++）**
- [ ] 实现任意算子的schedule优化，性能达到最新Tengine-lite的70%以上（难度：困难）

### **测试/编译类（C/C++）**
- [ ] 基于最新版本，优化测试代码/编译流程，优化使用体验（难度：一般）
- [ ] 基于最新版本，测试AutoKernel的算子性能, 提交到 `doc/benchmark.md`（难度：一般）

### **文档类（C/C++）**
- [ ] 基于最新版本，在现有的xxx.md 中修复 ≥ 3处文本错误/ ≥ 1处新内容增加。（难度：简单）

----
### 第三期悬赏任务说明：
做任务之前，可以先通过[AutoKernel初级教程](https://github.com/OAID/AutoKernel/tree/main/doc/tutorials)了解AutoKernel项目
1. 所有悬赏任务均以 Pull Request 形式**提交到 [AutoKernel项目](https://github.com/OAID/AutoKernel)**
2. 奖励标准：提交时间和任务完成质量，评选规则由 Tengine 开源委员会评选和决定；
3. 该期悬赏任务截止时间：2020.1.10；奖励和贡献者证书在 2020.1.31 统一发送；
4. 奖励发送&加入贡献者交流群，请添加Tengine小助手微信号：Tengine666 备注：任务
5. Tengine 开源委员会保留最终解释权。

lyuchuny3, 2020-12-11:
任务完成：基于最新版本，实现 maxpooling, avgpooling的移植，验证正确性（难度：简单）

任务完成：基于最新版本，实现 softmax的移植，验证正确性（难度：简单）

任务完成：基于最新版本，实现 fc的移植，验证正确性（难度：简单）

任务完成：基于最新版本，优化测试代码/编译流程，优化使用体验（难度：一般）

BUG1989, 2020-12-18:
任务完成：在现有的xxx.md 中修复 ≥ 3处文本错误/ ≥ 1处新内容增加（难度：简单）#502 #503 #504 #505 #506 #507