OAID/Tengine #496: Set ONNX model to acl_opencl failed

destinyzs, 2020-12-10:
I trained a Pytorch model, and convert it to ONNX model, but if i deploy it to rk3399 using `acl_opencl`, it will take the following error.

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  in validate_arguments src/core/CL/kernels/CLCol2ImKernel.cpp:52: Objects have different dimensions
Aborted
```

I have tried ONNX and tmfile, both will take this error, tmfile was converted using `./convert_model_to_tm`. But if i only use cpu, everything is ok... for both models.

Deploy code is like following

```
graph_t graph = create_graph(nullptr, "onnx", model_file.c_str());

// set_graph_device(graph, "acl_opencl");
int dims[] = {1, 1, img_h, img_w};
set_tensor_shape(input_tensor, dims, 4);
set_tensor_buffer(input_tensor, input_data, img_size * 4);

ret = run_graph(graph, 1);
```

BUG1989, 2021-02-04:
@destinyzs  we update the docs, compile, codes and examples of ACL implement #540 , please try it again, thanks.

closed 2021-02-04 by BUG1989

destinyzs, 2021-02-08:
@BUG1989  Thanks a lot for replaying, but i am using Tengine, not Tengine-lite... :disappointed_relieved: