OAID/Tengine #525: no serializer found for tengine

code-wangshuyi, 2021-01-06:
tengine-lite library version: 1.2-dev
Tengine: No available device.
no serializer found for tengine
Load model to graph failed(0).

运行的是example下的tm_retinaface

BUG1989, 2021-01-06:
@code-wangshuyi 请问是什么硬件平台和操作系统？谢谢。

code-wangshuyi, 2021-01-06:
arm32 hf

code-wangshuyi, 2021-01-06:
ubuntu 20.04 全志h2+

BUG1989, 2021-01-06:
板上编译还是交叉编译呢？

code-wangshuyi, 2021-01-06:
请问是我没有指定推理设备吗，我是通过交叉编译的

code-wangshuyi, 2021-01-06:
[image]

code-wangshuyi, 2021-01-06:
我改了源码中的cmakelist.txt，生成了静态库，我是用的opencv是opencv_world,这些有影响吗？另外我可以吧生成的静态库给到你，只要使用ubuntu 20.04 的交叉编译器就能复现我的错误了。
sudo apt-get install g++-arm-linux-gnueabihf

BUG1989, 2021-01-06:
参考 https://github.com/OAID/Tengine/blob/tengine-lite/toolchains/arm-linux-gnueabihf.toolchain.cmake 试试

BUG1989, 2021-01-06:
##### linux of arm-linux-gnueabihf toolchain
mkdir -p build-arm-linux-gnueabihf
pushd build-arm-linux-gnueabihf
cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/arm-linux-gnueabihf.toolchain.cmake ..
make -j$CPU_NUMS && make install
popd

code-wangshuyi, 2021-01-06:
稍等，马上反馈

code-wangshuyi, 2021-01-06:
还是不行[image]

code-wangshuyi, 2021-01-06:
我这边正在使用x86验证一下，看看行不行

code-wangshuyi, 2021-01-06:
x86是可以的，但是 orange pi  的 armbian 上面不行，是我交叉编译的问题吗？或者是我的工具链有问题？
[image]

BUG1989, 2021-01-06:
在板上本地编译可以吗

code-wangshuyi, 2021-01-06:
估计你得等到明天了，我这板子比较挫，明天我给你反馈吧

BUG1989, 2021-01-06:
我这边交叉编译 allwiner  V831是没问题的，别放弃啊

code-wangshuyi, 2021-01-06:
编译得慢/(ㄒoㄒ)/~~/(ㄒoㄒ)/~~/(ㄒoㄒ)/~~/(ㄒoㄒ)/~~/(ㄒoㄒ)/~~

code-wangshuyi, 2021-01-06:
我这边在板子上编译运行没有问题，是不是我是用交叉编译器设置的参数有问题

code-wangshuyi, 2021-01-07:
我这边知道原因了，CMAKELIST文件应该改错了，如果我想编译静态库，我都应该修改哪些文件呢？

BUG1989, 2021-01-10:
https://github.com/OAID/Tengine/blob/9359f5d4874f9fac8e5fb118950d65a4114886fd/src/CMakeLists.txt#L215 将这里的 SHARED 改为 STATIC

code-wangshuyi, 2021-01-13:
我当初也是这么干的，现在又重新试了一下，还是报这个错误
tengine-lite library version: 1.2-dev
Tengine: No available device.
no serializer found for tengine
Load model to graph failed(0).

BUG1989, 2021-05-02:
@code-wangshuyi 

Please try the latest version.

closed 2021-05-02 by BUG1989