OAID/Tengine #532: compile error with acl

idreamerhx, 2021-01-16:
rk3399 ubuntu 18

root@1e5b70eec222:/opt/Tengine/build# g++ --version
g++ (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0

`
export ACL_ROOT=/opt/ARM-software/arm_compute-v20.02-bin-linux

cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/arm64.native.gcc.toolchain.cmake \
-DTENGINE_ENABLE_ACL=ON \
-DTENGINE_ACL_LIB_DIRS=$ACL_ROOT/lib/linux-arm64-v8a-neon-cl \
-DCMAKE_CXX_FLAGS="-I$ACL_ROOT/include -I$ACL_ROOT -I$ACL_ROOT/arm_compute -I$ACL_ROOT/support" \
-DOpenCV_DIR=/opt/opencv-4.5.0/inst/lib/cmake/opencv4 \
..
`

make -j6

`
[ 85%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/wino_sgemm_4x16_A72.S.o
[ 85%] Building C object src/CMakeFiles/tengine-lite.dir/dev/acl/acl_allocator.c.o
[ 85%] Building CXX object src/CMakeFiles/tengine-lite.dir/dev/acl/acl_device.cpp.o
/opt/Tengine/src/dev/acl/acl_allocator.c:1543:17: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
     .allocate = acl_allocate
                 ^~~~~~~~~~~~
/opt/Tengine/src/dev/acl/acl_allocator.c:1543:17: note: (near initialization for 'acl_allocator.allocate')
In file included from /opt/Tengine/include/module.h:30:0,
                 from /opt/Tengine/include/tengine_ir.h:31,
                 from /opt/Tengine/src/dev/acl/acl_device.cpp:34:
/opt/Tengine/include/module.h:67:52: error: pasting ")" and "_t_" does not give a valid preprocessing token
     DECLARE_AUTO_INIT_FUNC(UNIQ_DUMMY_NAME(mod_init));     \
                                                    ^
/opt/Tengine/include/compiler.h:60:12: note: in definition of macro 'DECLARE_AUTO_INIT_FUNC'
     struct f##_t_                                               \
            ^
/opt/Tengine/include/nn_device.h:55:5: note: in expansion of macro 'REGISTER_MODULE_INIT_ARG'
     REGISTER_MODULE_INIT_ARG(MOD_DEVICE_LEVEL, "register_nn_device", ( module_init_func_t )register_nn_device, dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/src/dev/acl/acl_device.cpp:781:1: note: in expansion of macro 'REGISTER_NN_DEVICE'
 REGISTER_NN_DEVICE(&acl_dev.base);
 ^~~~~~~~~~~~~~~~~~
/opt/Tengine/include/module.h:67:52: error: pasting ")" and "_t_" does not give a valid preprocessing token
     DECLARE_AUTO_INIT_FUNC(UNIQ_DUMMY_NAME(mod_init));     \
                                                    ^
/opt/Tengine/include/compiler.h:62:9: note: in definition of macro 'DECLARE_AUTO_INIT_FUNC'

`

`
 ^
In file included from /opt/Tengine/include/module.h:30:0,
                 from /opt/Tengine/include/tengine_ir.h:31,
                 from /opt/Tengine/src/dev/acl/acl_device.cpp:34:
/opt/Tengine/include/compiler.h:60:15: error: variable 'auto_dummy_mod_init_781 _t_' has initializer but incomplete type
     struct f##_t_                                               \
               ^
/opt/Tengine/include/module.h:67:5: note: in expansion of macro 'DECLARE_AUTO_INIT_FUNC'
     DECLARE_AUTO_INIT_FUNC(UNIQ_DUMMY_NAME(mod_init));     \
     ^~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/include/nn_device.h:55:5: note: in expansion of macro 'REGISTER_MODULE_INIT_ARG'
     REGISTER_MODULE_INIT_ARG(MOD_DEVICE_LEVEL, "register_nn_device", ( module_init_func_t )register_nn_device, dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/src/dev/acl/acl_device.cpp:781:1: note: in expansion of macro 'REGISTER_NN_DEVICE'
 REGISTER_NN_DEVICE(&acl_dev.base);
 ^~~~~~~~~~~~~~~~~~
/opt/Tengine/include/compiler.h:62:12: error: expected '}' before '_t_'
         f##_t_(void)                                            \
            ^
/opt/Tengine/include/module.h:67:5: note: in expansion of macro 'DECLARE_AUTO_INIT_FUNC'
     DECLARE_AUTO_INIT_FUNC(UNIQ_DUMMY_NAME(mod_init));     \
     ^~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/include/nn_device.h:55:5: note: in expansion of macro 'REGISTER_MODULE_INIT_ARG'
     REGISTER_MODULE_INIT_ARG(MOD_DEVICE_LEVEL, "register_nn_device", ( module_init_func_t )register_nn_device, dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/src/dev/acl/acl_device.cpp:781:1: note: in expansion of macro 'REGISTER_NN_DEVICE'
 REGISTER_NN_DEVICE(&acl_dev.base);
 ^~~~~~~~~~~~~~~~~~
/opt/Tengine/include/compiler.h:62:12: error: expected ',' or ';' before '_t_'
         f##_t_(void)                                            \
            ^
/opt/Tengine/include/module.h:67:5: note: in expansion of macro 'DECLARE_AUTO_INIT_FUNC'
     DECLARE_AUTO_INIT_FUNC(UNIQ_DUMMY_NAME(mod_init));     \
     ^~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/include/nn_device.h:55:5: note: in expansion of macro 'REGISTER_MODULE_INIT_ARG'
     REGISTER_MODULE_INIT_ARG(MOD_DEVICE_LEVEL, "register_nn_device", ( module_init_func_t )register_nn_device, dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/src/dev/acl/acl_device.cpp:781:1: note: in expansion of macro 'REGISTER_NN_DEVICE'
 REGISTER_NN_DEVICE(&acl_dev.base);
 ^~~~~~~~~~~~~~~~~~
/opt/Tengine/include/compiler.h:66:5: error: expected declaration before '}' token
     };                                                          \
     ^
/opt/Tengine/include/module.h:67:5: note: in expansion of macro 'DECLARE_AUTO_INIT_FUNC'

`

/opt/Tengine/include/nn_device.h:55:5: note: in expansion of macro 'REGISTER_MODULE_INIT_ARG'
     REGISTER_MODULE_INIT_ARG(MOD_DEVICE_LEVEL, "register_nn_device", ( module_init_func_t )register_nn_device, dev)
     ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/Tengine/src/dev/acl/acl_device.cpp:781:1: note: in expansion of macro 'REGISTER_NN_DEVICE'
 REGISTER_NN_DEVICE(&acl_dev.base);
 ^~~~~~~~~~~~~~~~~~
src/CMakeFiles/tengine-lite.dir/build.make:8950: recipe for target 'src/CMakeFiles/tengine-lite.dir/dev/acl/acl_device.cpp.o' failed
make[2]: *** [src/CMakeFiles/tengine-lite.dir/dev/acl/acl_device.cpp.o] Error 1
CMakeFiles/Makefile2:85: recipe for target 'src/CMakeFiles/tengine-lite.dir/all' failed
make[1]: *** [src/CMakeFiles/tengine-lite.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

shiguang0910, 2021-01-26:
I feel the code problem of tengine are so much,who can solve this ,please!please!

BUG1989, 2021-02-03:
@idreamerhx and @shiguang0910  I have fixed this compile error in #540 depend on ACL v20.05, pls try it again.

closed 2021-02-04 by BUG1989