OAID/Tengine #544: Could not build Tengine on M1 MacBook Air

atinfinity, 2021-02-06:
I tried to use Tengine M1 MacBook Air.  
But, I could not build Tengine.

### System information (version)

- Tengine => Tengine Lite release v1.2 <https://github.com/OAID/Tengine/releases/tag/lite-v1.2>
- Operating System / Platform => macOS Big Sur 11.2
- Compiler => gcc 10.2.1(installed from homebrew)
- CPU: Apple M1

### Detailed description
#### CMake message

```
-- CMAKE_INSTALL_PREFIX = /Users/dandelion/dev/Tengine/build/install
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Checking whether C compiler has -isysroot
-- Checking whether C compiler has -isysroot - yes
-- Checking whether C compiler supports OSX deployment target flag
-- Checking whether C compiler supports OSX deployment target flag - yes
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/homebrew/bin/gcc-10 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking whether CXX compiler has -isysroot
-- Checking whether CXX compiler has -isysroot - yes
-- Checking whether CXX compiler supports OSX deployment target flag
-- Checking whether CXX compiler supports OSX deployment target flag - yes
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/homebrew/bin/g++-10 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- operation system is Darwin-20.3.0
-- other platform: Darwin
-- Now is Apple systems.
-- MAC OS COMPILER
-- The ASM compiler identification is GNU
-- Found assembler: /opt/homebrew/bin/gcc-10
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- TENGINE: OpenMP was found.
-- TENGINE: OpenMP support enabled.
-- TENGINE: 'OpenMP_C_FOUND' is TRUE.
-- TENGINE: 'TENGINE_LINKING_LIBS' is OpenMP::OpenMP_C.
CMake Warning at examples/CMakeLists.txt:59 (message):
  OpenCV not found, some examples won't be built

--
--
-- Infomation Summary:
--
-- CMake infomation:
--   - CMake version:              3.19.4
--   - CMake generator:            Unix Makefiles
--   - CMake building tools:       /usr/bin/make
--   - Target System:              Darwin
--   - Target CPU arch:            ARM
--   - Target CPU bus width:       64 Bit
--
-- tengine-lite toolchain infomation:
--   Cross compiling: FALSE
--   C/C++ compilier:
--     - C   standard version:     C99
--     - C   standard required:    TRUE
--     - C   standard extensions:  ON
--     - C   compilier version:    10.2.1
--     - C   compilier:            /opt/homebrew/bin/gcc-10
--     - C++ standard version:     C++11
--     - C++ standard required:    TRUE
--     - C++ standard extensions:  OFF
--     - C++ compilier version:    10.2.1
--     - C++ compilier:            /opt/homebrew/bin/g++-10
--   C/C++ compilier flags:
--     - C   compilier flags:
--     - C++ compilier flags:
--   OpenMP:
--     - OpenMP was found:         YES
--     - OpenMP version:           4.5
--
-- tengine-lite building infomation:
--   - Project source path is:     /Users/dandelion/dev/Tengine
--   - Project building path is:   /Users/dandelion/dev/Tengine/build
--
-- tengine-lite other infomation:
--   Package install path:         /Users/dandelion/dev/Tengine/build/install
--
-- Configuring done
-- Generating done
```

#### gcc message

```
[ 83%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s1p1_a72.S.o
[ 83%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0.S.o
/Users/dandelion/dev/Tengine/src/dev/cpu/op/aarch64/dw_k3s1p1_a72.S:61:1: error: unknown directive
.hidden dw_k3s1p1_a72
^
/Users/dandelion/dev/Tengine/src/dev/cpu/op/aarch64/dw_k3s1p1_a72.S:62:1: error: unknown directive
.type dw_k3s1p1_a72, %function
^
make[2]: *** [src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s1p1_a72.S.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/Users/dandelion/dev/Tengine/src/dev/cpu/op/aarch64/dw_k3s2p0.S:70:1: error: unknown directive
.hidden dw_k3s2p0
^
/Users/dandelion/dev/Tengine/src/dev/cpu/op/aarch64/dw_k3s2p0.S:71:1: error: unknown directive
.type dw_k3s2p0, %function
^
make[2]: *** [src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/dw_k3s2p0.S.o] Error 1
make[1]: *** [src/CMakeFiles/tengine-lite.dir/all] Error 2
make: *** [all] Error 2
```

### Steps to reproduce

```
$ git clone https://github.com/OAID/Tengine.git -b lite-v1.2
$ cd Tengine
$ mkdir build
$ cd build
$ cmake -D CMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=gcc-10 -DCMAKE_CXX_COMPILER=g++-10 .
$ make -j4
```

lyuchuny3, 2021-02-07:
@atinfinity 
I try to fix this problem at https://github.com/lyuchuny3/Tengine/commits/7a630606ad131f9044bfb6dcb928cb5b26361012
you can test if this problem can be solved.
I don't have the machine to test, please test it. 
Thanks.

atinfinity, 2021-02-07:
@lyuchuny3 Thank you for your help! I tried to build your code.
As a result, it occurs other problem.

```shell
[ 83%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/scale_neon.S.o
/Users/dandelion/dev/Tengine/src/dev/cpu/op/aarch64/scale_neon.S:148:1: error: unknown directive
.hidden scale_neon_bias
^
/Users/dandelion/dev/Tengine/src/dev/cpu/op/aarch64/scale_neon.S:149:1: error: unknown directive
.type scale_neon_bias, %function
^
make[2]: *** [src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/scale_neon.S.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 83%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/sgemm_4x16_deconv_a72.S.o
[ 84%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/sgemm_4x16_a72.S.o
make[1]: *** [src/CMakeFiles/tengine-lite.dir/all] Error 2
make: *** [all] Error 2
```

lyuchuny3, 2021-02-08:
@atinfinity 
try the lastest commit https://github.com/lyuchuny3/Tengine/tree/8ef5eebe4c0e740b07724fafc52846833beb2133

atinfinity, 2021-02-08:
@lyuchuny3 I tried your latest code.

```shell
[ 86%] Building ASM object src/CMakeFiles/tengine-lite.dir/dev/cpu/op/aarch64/wino_sgemm_4x16_A72.S.o
[ 86%] Linking C shared library libtengine-lite.dylib
Undefined symbols for architecture arm64:
  "_depthwise_k3s2p1_int8_a72", referenced from:
      _conv_dw_int8_direct._omp_fn.0 in conv_dw_kernel_int8_arm.c.o
ld: symbol(s) not found for architecture arm64
collect2: error: ld returned 1 exit status
make[2]: *** [src/libtengine-lite.dylib] Error 1
make[1]: *** [src/CMakeFiles/tengine-lite.dir/all] Error 2
make: *** [all] Error 2
```

kalcohol, 2021-03-08:
it seems we should push our company to buy at least one m1 to fix this issue...

ctcx, 2022-07-23:
快一年了，请问公司给你配M1的苹果电脑了吗？现在Tengine支持 M1 Macos 了吗？

> 看来我们应该推动我们公司至少购买一台 m1 来解决这个问题......

ctcx, 2022-07-23:
clangclang: : warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_hcl_arm.c:60:13: warning: address of function 'conv_hcl_set_shared_mem' will always evaluate to 'true' [-Wpointer-bool-conversion]
        if (conv_hcl_set_shared_mem && exec_node->shared_mem_size < exec_graph->shared_mem_size)
            ^~~~~~~~~~~~~~~~~~~~~~~ ~~
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_hcl_arm.c:60:13: note: prefix with the address-of operator to silence this warning
        if (conv_hcl_set_shared_mem && exec_node->shared_mem_size < exec_graph->shared_mem_size)
            ^
            &
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_hcl_arm.c:68:13: warning: address of function 'conv_hcl_set_shared_pack4_mem' will always evaluate to 'true' [-Wpointer-bool-conversion]
        if (conv_hcl_set_shared_pack4_mem && exec_node->shared_pack4_mem_size < exec_graph->shared_pack4_mem_size)
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_hcl_arm.c:68:13: note: prefix with the address-of operator to silence this warning
        if (conv_hcl_set_shared_pack4_mem && exec_node->shared_pack4_mem_size < exec_graph->shared_pack4_mem_size)
            ^
            &
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_hcl_arm.c:109:13: warning: address of function 'int8_conv_hcl_set_shared_mem' will always evaluate to 'true' [-Wpointer-bool-conversion]
        if (int8_conv_hcl_set_shared_mem && exec_node->shared_mem_size < exec_graph->shared_mem_size)
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_hcl_arm.c:109:13: note: prefix with the address-of operator to silence this warning
        if (int8_conv_hcl_set_shared_mem && exec_node->shared_mem_size < exec_graph->shared_mem_size)
            ^
            &
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/conv/cortex-a/conv_kernel_int8_arm.c:2079:19: warning: result of comparison of constant 2147483648 with expression of type 'int' is always false [-Wtautological-constant-out-of-range-compare]
        if (fix_q == (1l << 31))
            ~~~~~ ^  ~~~~~~~~~~
3 warnings generated.
[ 29%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/conv/cortex-a/wino_conv_kernel_1_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
[ 29%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/conv/cortex-a/wino_conv_kernel_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
1 warning generated.
[ 29%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/eltwise/cortex-a/eltwise_hcl_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
[ 29%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/eltwise/cortex-a/eltwise_kernel_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
[ 29%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/elu/cortex-a/elu_hcl_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
[ 30%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/elu/cortex-a/elu_kernel_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
[ 30%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/fc/cortex-a/fc_hcl_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
/Users/tony/Documents/Work/ai/Tengine/source/device/cpu/op/fc/cortex-a/fc_hcl_arm.c:45:10: fatal error: 'cortex_a/fc_kernel_fp16_arm82.h' file not found
#include "cortex_a/fc_kernel_fp16_arm82.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [source/CMakeFiles/tengine-lite.dir/device/cpu/op/fc/cortex-a/fc_hcl_arm.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 30%] Building C object source/CMakeFiles/tengine-lite.dir/device/cpu/op/fc/cortex-a/fc_kernel_arm.c.o
clang: warning: argument unused during compilation: '-Xclang -fopenmp' [-Wunused-command-line-argument]
make[1]: *** [source/CMakeFiles/tengine-lite.dir/all] Error 2
make: *** [all] Error 2