OAID/Tengine #551: No serializer found for tengine for arm cortex A7

gihyunkim, 2021-02-16:
Hi, I have been through some problems below. and hope you can help me out from this error.

==========================
tengine-lite library version: 1.2-dev
Tengine: No available device.
no serializer found for tengine
Create graph failed.
errno: 0
==========================
This is my cross compile environment
- ubuntu 18.04 with virtualBox
- toolchain : arm-linux-gnueabi-
==========================
I tried your some examples and it worked well on above environment with no cross compile.
The build and make is successful but if I load it to device, the error happens. 
I added the flags -mfpu=neon-vfpv4 -mfloat-abi=softfp -march=armv7-a as well, actually it had been used already.

Please help me to figure it out this problem.
Thank you.

gihyunkim, 2021-02-16:
I did get rid of the error "No Serializer found for tengine" by adding src codes in Serializer folder from your structure to our structure But I get error "Tengine: No available device" still.
For running our code with tengine in Arm environment, what am I supposed to do? I'm confused which source file needs to be included or not. Plz Let me know thank you

BUG1989, 2021-02-17:
Hi @gihyunkim which A7 device did you try it?

gihyunkim, 2021-02-17:
Hi Thank you for responding. I've tried AllWinner A40 device.

gihyunkim, 2021-02-18:
> Hi @gihyunkim which A7 device havd you tried it?
@BUG1989 
I've drawed files in /src/cpu/ as well this time because I supposed it cause "no device support". 
And I got this error below.

deconv_kernel_arm.c:(.text+0x788): undefined reference to `sgemm_4x12_deconv_a17'
deconv_kernel_arm.c:(.text+0x7b0): undefined reference to `sgemm_4x12_deconv_a17'
deconv_kernel_arm.c:(.text+0x1c0a): undefined reference to `sgemm_4x4_deconv_a17'
deconv_kernel_arm.c:(.text+0x1ca0): undefined reference to `sgemm_4x4_deconv_a17'
deconv_kernel_arm.c:(.text+0x1d16): undefined reference to `sgemm_4x4_deconv_a17'
...

I might guess it happens because I assembled all files in src/lib,  src/dev to one folder.(Because I don't know which files i need)
I hope you can give me an answer to fix this error, Thank you.

kalcohol, 2021-03-08:
> deconv_kernel_arm.c:(.text+0x788): undefined reference to `sgemm_4x12_deconv_a17' deconv_kernel_arm.c:(.text+0x7b0): undefined reference to `sgemm_4x12_deconv_a17' [...]

Sorry for the late reply, src/CMakeList.cmake show how conditions Tengine based and how to gather the source files for different target. Such as ARM, X86 or MIPS.