OAID/Tengine #555: 关于caffmodel转换为tmfile在RK3399上调用ACL出错的问题

liangX-box, 2021-02-22:
hello：
我将我自己的retinaface-caffemodel转换为tmfile后，在RK3399 arm平台上运行tengine graph, cpu版的没问题，但是调用acl运行的时候出现如下错误：
terminate called after throwing an instance of "std::runtime_error"
what(): in calculate_concatenate_shape ./arm_compute/core/utils/misc/ShapeCalculator.h:1332:out_shape[i] != shape[i]
Aborted (core dumped)
好像caffemodel转换为tmfile后都会出现上述问题。

BUG1989, 2021-02-24:
#559 已解决该问题

closed 2021-02-24 by BUG1989

ideafold, 2021-12-27:
目测没彻底解决哈。

tengine-lite library version: 1.5-dev
before prerun_graph_multithread.
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
ACL initialized
terminate called after throwing an instance of 'std::runtime_error'
  what():  in calculate_concatenate_shape ./arm_compute/core/utils/misc/ShapeCalculator.h:1332: out_shape[i] != shape[i]
Aborted (core dumped)