OAID/Tengine #564: when run ’tm_classification_timvx“  ，Failed to open device, errno=No such file or directory

wuqiangch, 2021-02-24:
I follow 'npu_tim-vx_user_manual.md' to install tengine.  
./tm_benchmark 
Tengine plugin allocator TIMVX is registered.
loop_counts = 1
num_threads = 1
power       = 0
affinity    = 255
tengine-lite library version: 1.2-dev
     squeezenet_v1.1  min =   56.50 ms   max =   56.50 ms   avg =   56.50 ms
         mobilenetv1  min =  104.54 ms   max =  104.54 ms   avg =  104.54 ms
         mobilenetv2  min =   91.66 ms   max =   91.66 ms   avg =   91.66 ms
         mobilenetv3  min =   62.15 ms   max =   62.15 ms   avg =   62.15 ms
        shufflenetv2  min =   30.48 ms   max =   30.48 ms   avg =   30.48 ms
            resnet18  min =  163.78 ms   max =  163.78 ms   avg =  163.78 ms
            resnet50  min =  490.07 ms   max =  490.07 ms   avg =  490.07 ms
           googlenet  min =  200.13 ms   max =  200.13 ms   avg =  200.13 ms
         inceptionv3  min =  802.00 ms   max =  802.00 ms   avg =  802.00 ms
               vgg16  min =  872.98 ms   max =  872.98 ms   avg =  872.98 ms
                mssd  min =  207.17 ms   max =  207.17 ms   avg =  207.17 ms
          retinaface  min =   28.91 ms   max =   28.91 ms   avg =   28.91 ms
         yolov3_tiny  min =  224.99 ms   max =  224.99 ms   avg =  224.99 ms
      mobilefacenets  min =   44.56 ms   max =   44.56 ms   avg =   44.56 ms
ALL TEST DONE
./tm_classification_timvx -m ./models/mobilenet_uint8.tmfile -i ./models/face1.jpg -g 224,224 -s 0.017,0.017,0.017 -w 104.007,116.669,122.679 -r 500
Tengine plugin allocator TIMVX is registered.
tengine-lite library version: 1.2-dev
TIM-VX prerun.
[ 1] Failed to open device: No such file or directory, Try again...
[ 2] Failed to open device: No such file or directory, Try again...
[ 3] Failed to open device: No such file or directory, Try again...
[ 4] Failed to open device: No such file or directory, Try again...
[ 5] _OpenDevice(1130): FATAL: Failed to open device, errno=No such file or directory.
Can‘t find the npu?

BUG1989, 2021-02-24:
@wuqiangch 请问您是使用的 Khadas VIM3 的板子吗？如果是的话，这个问题大概率是由于 Khadas VIM3 预装的 libtengine-lite.so 和 galcore.ko 冲突导致的。大致解决思路如下：

1. 通过环境变量来指定新编译好的 libtengine-lite.so 的路径；
2. 按照[Tengine Lite with VeriSilicon TIM-VX User Manual](https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual.md)，替换掉以下库文件：
```
3rdparty/tim-vx/lib/
├── libArchModelSw.so
├── libCLC.so
├── libGAL.so
├── libNNArchPerf.so
├── libOpenVX.so
├── libOpenVXU.so
└── libVSC.so
```
3. 更新 galcore.ko。

wuqiangch, 2021-02-25:
@BUG1989 我用的就是hadas VIM3 的板子。我试了还是不行。因为我重新刷了系统，我对这个NPU不是很了解。是不是还要装NPU驱动之类的，就像NVIDIA的GPU一样需要装驱动。
我用这个命令，sudo dmesg |grep npu
[    4.574021] input: gpio_keypad as /devices/platform/gpio_keypad/input/input0
[    4.725820] input: aml_keypad as /devices/platform/ff808040.rc/input/input1
[    5.889111] input: cec_input as /devices/virtual/input/input2
[    7.445813] input: adc_keypad as /devices/platform/adc_keypad/input/input3
[    8.294730] input: PixArt Dell MS116 USB Optical Mouse as /devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.1/1-1.1:1.0/0003:413C:301A.0001/input/input4
[    8.305696] hid-generic 0003:413C:301A.0001: input,hidraw0: USB HID v1.11 Mouse [PixArt Dell MS116 USB Optical Mouse] on usb-xhci-hcd.0.auto-1.1/input0
[    8.552295] input: DELL Dell USB Entry Keyboard as /devices/platform/ff500000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1.2/1-1.2:1.0/0003:413C:2107.0002/input/input5
[    8.619315] hid-generic 0003:413C:2107.0002: input,hidraw1: USB HID v1.11 Keyboard [DELL Dell USB Entry Keyboard] on usb-xhci-hcd.0.auto-1.2/input0
没有发现和npu相关的。

BUG1989, 2021-02-25:
加我 QQ 吧，我来看看。

BUG1989, 2021-02-28:
已经线下进行沟通

closed 2021-02-28 by BUG1989

wuqiangch, 2021-03-03:
已经解决了，还是因为是系统的原因。
一定要选择最新的系统
网址：https://docs.khadas.com/firmware/Vim3Firmware.html
下载最新版系统：
“VIM3_Ubuntu-gnome-focal_Linux-4.9_arm64_EMMC_V0.9-20200530(Latest release)”这个版本
把系统烧进去就可以用了。
进入系统显示如下：
[image]
查看npu的版本：
[image]
查看npu的驱动版本如下：
[image]