OAID/Tengine #579: Not found kernel "prelu" E [compute_node:354]Create node[1] PRELU fail

wuqiangch, 2021-03-11:
我的模型里面prelu算法，量化之后，在vim3的npu上面运行出现以上问题。我又把prelu替换为relu，又出现其他错误：
kernel_selector:906]Not found kernel "add"
E [compute_node:354]Create node[6] ADD fail

wuqiangch, 2021-03-13:
代码更新解决了Add的问题，但是prelu仍然不支持。带有prelu的算子提示一下错误：  
ERROR: Failed to compile vx shader. (error: FFFFFFFF)
E [_gpu_register:476]Build program fail.
E [vsi_nn_kernel_create_node:631]Register client kernel com.vivantecorp.extension.evis.prelu_U8F16toU8_2D fail with -1.

(10:0) : error : Error(0,10) : Cannot find the header file cl_viv_vx_ext.h.
(200:0) : error : Error(0,200) : Cannot find the header file cl_viv_vx_ext.h.
(203:0) : error : syntax error at 'VXC_512Bits'

BUG1989, 2021-03-15:
需要将 cl_viv_vx_ext.h 拷贝到执行文件的路径下，为什么呢？prelu 的算子是在线编译生成的.......

closed 2021-03-15 by BUG1989