OAID/Tengine #585: does TIMVX support int models?

stevenwudi, 2021-03-17:
According to Khadas (https://docs.khadas.com/vim3/HowToConvertToUseNPU.html)，the quantization method is "dyanmic_xied_point-i8".

But  “The TIM-VX NPU backend needs the uint8 tmfile as it's input model file, you can quantize the tmfile from float32 to uint8 from here.”，（https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual.md)

So I am kinda confused, does TIMVX support int8 model or uint8 model? Or both? 
Thanks in advance for the clarification.

Adhithya-tech, 2021-03-25:
How to convert .pb/.pt file to tmfile? Or how do you execute .c file in tengine? I'm having difficulty downloading the tengine env. and 'import tengine'
@stevenwudi

BUG1989, 2021-03-27:
@stevenwudi Only support uint8 model.

closed 2021-05-08 by stevenwudi