OAID/Tengine #608: quant_tool_uint8 failed

XinnWang, 2021-04-01:
Hi there,

i want to convert my onnx model to uint8 tmfile. my onnx model has 1 x 1 x 350 x 640 (nchw) input.

First, i use the tm_convert_tool to convert onnx to tmfile successfully.
After that, i use the quant_tool_uint8 to convert tmfile to uint8 tmfile, but failed with segmentation fault (core dumped).

The README.md inside Tengine/tools/quantize only shows how to transfer a model with 3-channel image input, but my model has 1-channel image input.

i tried some parameters but all failed at segmentation fault (core dumped).  Any helps or suggestions? Thanks.

stevenwudi, 2021-04-09:
Me two, when I converted yolov5 1*3*320*320, also generated segmentation fault (core dumped).
BTW, for 1*3*640*640 images, it works fine (yolov5)....

@BUG1989  Any chance of opensource this quantization tool so that the communities might be able to help?

Looking forward to it.

BUG1989, 2021-04-10:
The quantization tool will opensource.

stevenwudi, 2021-04-13:
> Hi there,
>
> i want to convert my onnx model to uint8 tmfile. my onnx model has 1 x 1 x 350 x 640 (nchw) input.
>
> First, i use the tm_convert_tool to convert onnx to tmfile successfully.
> After that, i [...]

Hey, in your scenario, you could change the onnx input to " 1 x 1 x 640x 640" (during training, you can use letter box for padding"

stevenwudi, 2021-04-13:
> Me two, when I converted yolov5 1x3x320x320, also generated segmentation fault (core dumped).
> BTW, for 1x3x640x640 images, it works fine (yolov5)....
> 
> @BUG1989 Any chance of opensource this quantization tool so that the communities might be able to help?
> 
> Looking forward to it.

@BUG1989 has successfully updated a quant_tool_uint8 version (time stamp: Apr  8 2021) to allow 1x3x320x320, for yolov5 (which now runs 20ms, ~50 fps) :
the command will be sth. like

`./quant_tool_uint8_320 -m yolov5s_320_opt.tmfile -i /coco  -o yolov5s_320_opt_uint8.tmfile -g 12,160,160 -w 0,0,0 -s 0.003921,0.003921,0.003921 -c 0 -k 1 -t 2 -y 320`
`-y 320` is the new input for image size.

BUG1989, 2021-06-09:
quant_tool_uint8 is open.

closed 2021-06-09 by BUG1989