OAID/Tengine #648: [Crash] quant_tool_uint8

qinhj, 2021-05-07:
Case 1: (reproduce possibility ~10% in my case)
$ ./quant_tool_uint8 -m $MODELS/yolov5s.v5.tmfile -i $IMAGES -o ../../build/install/yolov5s_uint8.tmfile -g 12,320,320 -w 0,0,0 -s 0.004,0.004,0.004 -c 0 -k 1 -y 640 -t 2

---- Tengine Post Training Quantization Tool ---- 

Version     : v1.1, 23:50:51 Apr 29 2021
Status      : uint8, per-layer, asymmetric
Input model : /media/sf_Workshop/Models/coco_nc80/yolov5s.v5.tmfile
Output model: ../../build/install/yolov5s_uint8.tmfile
Calib images: /media/sf_Workshop/Dataset/Yolo3/calibration/
Scale file  : NULL
Algorithm   : MIN MAX
Dims        : 12 320 320
Mean        : 0.000 0.000 0.000
Scale       : 0.004 0.004 0.004
BGR2RGB     : ON
Center crop : OFF
Letter box  : 640 779
YOLOv5 focus: ON
Thread num  : 2

[Quant Tools Info]: Step 0, load FP32 tmfile.
[Quant Tools Info]: Step 0, load FP32 tmfile done.
[Quant Tools Info]: Step 0, load calibration image files.
[Quant Tools Info]: Step 0, load calibration image files done, image num is 11.
[Quant Tools Info]: Step 1, find original calibration table.
[Quant Tools Info]: Step 1, images 00011 / 00011
[Quant Tools Info]: Step 1, find original calibration table done, output ./table_minmax.scale
[Quant Tools Info]: Thread 2, image nums 11, total time 3654.02 ms, avg time 332.18 ms
munmap_chunk(): invalid pointer
已放弃 (核心已转储)

Case 2: same as Case 1, except the crash info "double free"

qinhj, 2021-05-07:
It seems that the crash is caused by the option "-y 640", which results in the unexpected/incorrect "Letter box : 640 779".
In normal cases, it could be "Letter box : 640 0". 
Maybe some related variables are not initialed, or did I miss some thing?

BUG1989, 2021-05-09:
` -g 12,320,320 -w 0,0,0 -s 0.004,0.004,0.004 -c 0 -k 1 -y 640,640 -t 2`

@qinhj   The quant tool support asymmetric letter box, please try it again.

closed 2021-05-13 by qinhj