OAID/Tengine #664: Tengine-lite 1.4 run segmentation fault on VIM3

huoshuai-dot, 2021-05-13:
1、NPU_TIM_VX教程里面的log不是基于最新1.4版本的libtengine，想知道官方是否在vim3上跑过这个case？
2、我自己run这个case的时候 报段错误 发现是在prerun_graph_multithread（）这个函数segmentation fault，注释掉这个函数 可以跑下去 但是 结果明显不对 想知道这个可能存在的问题是什么？

BUG1989, 2021-05-18:
你是看的 https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual.md 来实验的吗？

huoshuai-dot, 2021-05-20:
> 你是看的 https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual.md 来实验的吗？
是的呀 x86仿真的没啥问题 在vim3上跑不通 不大明白啥原因 稍微有点区别就是第三部分 libtengine-lite.so这个路径跟文档里面的不是完全一致 不过 感觉不是这些问题

zylo117, 2021-05-20:
这个问题我也遇到过，是因为文档可以没讲清楚，有一步非常重要，就是要用新版的libOpenVX.1.3。
临时解决办法就是加个环境变量，指向这个版本的libOpenVX
`export LD_LIBRARY_PATH=/path/to/Tengine/3rdparty/tim-vx/lib/aarch64`

kalcohol, 2021-05-20:
> 这个问题我也遇到过，可能是因为文档没讲清楚，这一步非常重要，就是要用新版的libOpenVX.1.3。
> 临时解决办法就是加个环境变量，指向这个版本的libOpenVX
> `export LD_LIBRARY_PATH=/path/to/Tengine/3rdparty/tim-vx/lib/aarch64`

是的，当板子上有其他 tengine 的库，或多版本的 lib 时会有这个问题。
建议把下载到的库进行替换，替换到系统文件处，这样一劳永逸。
apt remove tengine-libs 移除 vim3 集成的 tengine

closed 2021-06-09 by BUG1989