OAID/Tengine #670: keras版本yolov3算法训练出来的模型经过onnx中间格式转为tmfile无法使用

Master-cai, 2021-05-14:
- keras版本yolov3算法训练出来的模型经过onnx中间格式转为tmfile无法使用。
使用的yolov3算法代码为 https://github.com/qqwweee/keras-yolo3 的代码，然后将其训练出来的.h5模型转为ONNX格式，再转换为tmfile格式文件之后无法在Tengine框架中使用（使用的测试代码为example中给出的tm_yolov4_tiny.cpp https://github.com/OAID/Tengine/blob/tengine-lite/examples/tm_yolov4_tiny.cpp）。报错信息为：
[image]
- 但是使用yolov3给出的darknet示例模型（weights和cfg文件）直接转换为tmfile之后，可以配合tm_yolov4_tiny.cpp运行。因此推测问题出在
所使用的设备为：EAIDK610，转换tmfile所使用的工具均为在线转换工具
- yolov3算法主要依赖版本为：keras=2.2.4 tensorflow=1.14 
- 相关文件放在这个网盘里了（包括：keras模型转onnx模型代码，YOLOV3示例darknet模型（.weights和.cfg文件），两个keras版本yolov3模型文件（一个模型是coco80类数据集训练的，另一个是识别person和hat两类的模型）,一个出问题的tmfile（person_hat模型转化的）以及其中间模型onnx文件） https://1drv.ms/u/s!Ak05bBUBFYiksfUdwh9qzx5hBH9BFw?e=vOjj9I

qinhj, 2021-05-18:
$ ./install/bin/convert_tool -f onnx -m hat.onnx -o hat.tmfile
---- Tengine Convert Tool ---- 

Version     : v1.0, 14:35:53 May 18 2021
Status      : float32
conv2d_1 attr.name:auto_pad:SAME_UPPER todo implement.
conv2d_3 attr.name:auto_pad:SAME_UPPER todo implement.
conv2d_4 attr.name:auto_pad:SAME_UPPER todo implement.
conv2d_6 attr.name:auto_pad:SAME_UPPER todo implement.
conv2d_7 attr.name:auto_pad:SAME_UPPER todo implement.
conv2d_8 attr.name:auto_pad:SAME_UPPER todo implement.
...

建议改用pytorch版本的yolov3，或手改hat.onnx文件~

Master-cai, 2021-05-18:
> $ ./install/bin/convert_tool -f onnx -m hat.onnx -o hat.tmfile
> ---- Tengine Convert Tool ----
>
> Version : v1.0, 14:35:53 May 18 2021
> Status : float32
> conv2d_1 attr.name:auto_pad:SAME_UPPER todo [...]

这个是什么意思呀，是因为原本的模型原因吗

qinhj, 2021-05-19:
> 这个是什么意思呀，是因为原本的模型原因吗

使用官方转换工具 https://github.com/OAID/Tengine-Convert-Tools 进行转换

closed 2021-10-17 by Master-cai