OAID/Tengine #698: NXP iMX8 Plus

Hank880223, 2021-05-27:
請問作者在建立 NXP iMX8 Plus 的 BSP 環境是幾版的
我下載 Tengine Lite release v1.4 for NXP 的版本，照著 README.md 進行編譯時，沒有辦法直接編譯成功，
sample/console/tengine/linux-arm64-v8a/lib/libtim-vx.so 在編譯時會提示缺少部分 lib，
後來將 TimVX 裡面的某些 lib 加進去後可以編譯成功，但是在板子上執行時還是會執行失敗

[image]

想請問作者在建立 NXP iMX8 Plus 的環境時是建立哪一個版本的，還是在建立時有添加某些關於 NPU 的編譯文件

謝謝

kalcohol, 2021-05-27:
@Hank880223 请看一下 TIM-VX 文档的 iMX8MP 部分，编译过程做了一些修改，可以直接拉取最新代码；
文档链接：https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual_zh.md
此外 FAQ 部分有讲怎么查看 Galcore 的版本，我这边看到的版本是[6.4.3.p1.305572]；列表的文件都是在 /usr/lib/ 下面的，具体必要的有：
libGAL.so
libNNGPUBinary.so
libOpenCL.so
libOpenVXU.so
libVSC.so
libCLC.so
libNNArchPerf.so
libNNVXCBinary.so
libOpenVX.so
libOvx12VXCBinary.so
libarchmodelSw.so
此外，我们是直接从 NXP 处拿到的 EVK，还不清楚怎么烧写 BSP，拿到的板子默认就是这样的，如果您知道怎么检查 BSP 版本，可以告诉我们。uname 打印是
Linux imx8mpevk 5.4.70-2.3.0+g4f2631b022d8 #1 SMP PREEMPT Wed Dec 16 11:32:01 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
后面我们会更新一个版本，fix bugs 和更多的 op。

Hank880223, 2021-05-28:
您好，我將 BSP 版本更換成 5.4.70 後，已經可以成功在板子上執行了。
不好意思，我這邊目前也沒有辦法依靠指令的方式得知目前使用的 BSP 版本是多少，我是透過 NXP 的文件步驟去進行環境編譯和燒錄 SD card，並以 SD card 進行開機，選擇要燒錄的版本是透過這個網址查詢https://source.codeaurora.org/external/imx/imx-manifest

[image]
[image]
[image]

如有找到如何檢查 BSP 版本會再跟您通知

目前可以成功測試 yolov3_uint8.tmfile 可是換成以 yolov3 為 base，自己的模型後沒有辦法在 NPU 上正常執行，想請問作者在NPU 上目前 yolov3 可以使用的 op 有哪些，我在PC端測試自己的模型是可以正常動作的，換到 NXP iMX8M Plus 後會出現以下錯誤
[image]

BUG1989, 2021-05-28:
@Hank880223 如果方便的话，可以在 QQ 群里进行进一步讨论。

Hank880223, 2021-05-28:
請問QQ群裡面的那個問題是要回答什麼嗎?

BUG1989, 2021-05-28:
> 請問QQ群裡面的那個問題是要回答什麼嗎?

CNN

Hank880223, 2021-05-28:
您好，已經申請加群

BUG1989, 2021-05-28:
> 您好，已經申請加群

并没有收到入群申请……你 QQ 号多少我拉你进群

Hank880223, 2021-05-28:
180064834

Hank880223, 2021-05-31:
經由測試後 Tengine Lite release v1.4 for NXP 裡面的版本好像還不支援 yolov3 mish 的 activate function，我將我的模型的 activate function 替換成 leaky 後可以正常執行，但是在 NPU 上執行會有精確度的問題，想請問作者是否在 yolov3 上是否已經有將 convolution 的 group op 添加到 NPU 上

以下是我執行 tm_yolov3_uint8 執行檔的結果
[image]
[image]

換成 Tengine Lite release v1.4 for NXP 裡面 for TIMVX 的版本後，會造成與 tm_yolov3_uint8 執行的結果不同
[image]

以下是我的 tmfile 和測試圖片 
[test.zip](https://github.com/OAID/Tengine/files/6568518/test.zip)

以下是我的模型參數
net_w = 256
net_h = 256
classes = 1
biases[18] = {5,  7,   9, 12,  16, 20,  29, 24,  23, 37,  29, 72,  50, 48,  60,101, 116,147};

fengwang, 2022-04-06:
> @Hank880223 如果方便的话，可以在 QQ 群里进行进一步讨论。

请问除了 QQ 群还有别的方式么？
海外用户 QQ 不能注册也不能登陆了