OAID/Tengine #699: 请问可以直接training tmfile出来吗?

b51ak47, 2021-05-27:
请问可以直接training tmfile出来吗? 因为tengine-convert-tool covert 会有error

tengine-lite library version: 1.4-dev
Get input tensor failed
[image]

或是有例子能training出下面tmfile 呢?
[image]

b51ak47, 2021-06-01:
或是想请问版主 能不能提供example 资料夹 那些 training module 的 source code 让我可以自己train看看呢?
拜托大神~

kingjames1288, 2021-06-09:
请问你找到了可用的training module 来训练自己的数据集了吗

b51ak47, 2021-06-11:
我现在就是希望能跟Tengine 要到可以run 的 training module 来跑我的资料集阿! 不管是python 的还是 其他的程式语言都好....

BUG1989, 2021-07-05:
tmfile 都是通过已有的训练框架（例如 PyTorch、MXNet）公开的模型转换过来的。