OAID/Tengine #763: Tengine-Master 分支 Mobilenet_SSD 物体检测无法正常运行

iMacroC, 2021-06-24:
Master分支编译出来的例子 无法正常检测，但是Lite分支编译出来的可以正常检测。
[image]
（Master分支下的Mobilenet_ssd例子）
[image]
（Lite分支下的tm_mobilenet_ssd例子）

用Python版本写出来的也是有同样的问题
[image]
（左边是Master分支下的Mobilenet_ssd例子，右边是自己用Python写出来的）

Master分支下的Mobilenet_ssd例子执行后 似乎把所有的物体都输出出来了，Score甚至577%
[image]

是Master分支编译出来的动态库有问题吗？使用libtengine-lite.so就没有问题，模型都是一样的，使用的mobilenet_ssd.tmfile。

iMacroC, 2021-06-24:
似乎Pytengine并不支持Tengine-Lite，import直接报错了。
[image]

BUG1989, 2021-06-25:
请使用 tengine-lite 分支

SwaggyMacro, 2021-06-25:
> 请使用 tengine-lite 分支

请问tengine-lite有相应的python-api支持吗

SwaggyMacro, 2021-06-25:
> 请使用 tengine-lite 分支

@BUG1989 请问tengine-lite有相应的python-api支持吗

BUG1989, 2021-06-25:
已经更新了

SwaggyMacro, 2021-06-25:
> 已经更新了

@BUG1989 谢谢，测试正常了

BUG1989, 2021-06-25:
@SwaggyMacro 谢谢反馈结果。

closed 2021-06-25 by BUG1989

SwaggyMacro, 2021-06-25:
> @SwaggyMacro 谢谢反馈结果。

@BUG1989 但是每次识别都有一个sofa score有143% 不过坐标在图片外 没有啥影响
[image]

SwaggyMacro, 2021-06-25:
[image]