OAID/Tengine #775: 官方TIM-VX例子在RV1109上运行错误

ChungTak, 2021-06-29:
#### 问题描述
官方TIM-VX例子在RV1109上运行，只有tm_classification_timvx能运行，其他都报错

1) yolov4_tiny_timvx
` ./tm_yolov4_tiny_timvx -m /userdata/yolov4-tiny_uint8.tmfile -i /userdata/ssd_dog.jpg`
输出
tengine-lite library version: 1.4-dev  
W [op_optimize:580]stride slice copy tensor.
W [op_optimize:580]stride slice copy tensor.
W [op_optimize:580]stride slice copy tensor.
Tengine Fatal: Pre-run subgraph(0) on TIMVX failed.
Tengine: Scheduler(sync) prerun failed.
Prerun multithread graph failed.
2) yolov5s_timvx
` ./tm_yolov5s_timvx  -m /userdata/yolov5s_uint8.tmfile -i /userdata/ssd_dog.jpg -r 1`
输出
tengine-lite library version: 1.4-dev
Tengine Fatal: Pre-run subgraph(0) on TIMVX failed.
Tengine: Scheduler(sync) prerun failed.
Prerun multithread graph failed.
3) hrnet_timvx
` ./tm_hrnet_timvx -m /userdata/hrnet_uint8.tmfile -i /userdata/runing.jpg -r 1 -t 1`
输出
tengine-lite library version: 1.4-dev
Segmentation fault
#### 运行环境
 RV1109开发板,SDK_2.1.0
#### 重现过程
1) git获取最新源代码 2021-06-28 
2) 根据 [Tengine Lite with VeriSilicon TIM-VX User Manual](https://github.com/OAID/Tengine/blob/91b7ca4aed33b0591d5858ffea216825d788b10f/doc/npu_tim-vx_user_manual_zh.md)编译成功,把install 和opencv依赖部署上rv1109设备
3) 下载官方模型 从 [GoogleDriver](https://drive.google.com/drive/folders/1hunePCa0x_R-Txv7kWqgx02uTCH3QWdS?usp=sharing) 下载的
4) 在1109设备上直接运行example

BUG1989, 2021-06-29:
@ChungTak 请问 3rdparty 是使用的网盘里面的吗？
[image]

ChungTak, 2021-06-29:
> 3rdparty

不是～～全部依赖都是从源代码编译的～～ 我试试网盘这个

ChungTak, 2021-06-29:
@BUG1989   3rdparty使用网盘的yolo4和5没问题了，但tm_hrnet_timvx 还是直接报  Segmentation fault (core dumped)

BUG1989, 2021-06-30:
@ChungTak 好的，我们这边在 RV1126 上验证下 hrnet 的问题

BUG1989, 2021-06-30:
目前我们观察到 rv1126 运行 hrnet 会存在内存不足的情况。

ChungTak, 2021-06-30:
噢，那估计1109我是这个问题

closed 2021-07-02 by ChungTak