OAID/Tengine #781: Tengine Fatal: Pre-run subgraph(0) on TIMVX failed.

AkkiSony, 2021-06-29:
I want to use Tengine to convert my darknet yolov3 model to NPU compatible format and run it on khadasVIM3. I have modified the paramaters in the .cpp file as follows:

**num_classes = 3;
coconames[3] = {"apple","phone","water"};**

I did not modify anything else other than these. Later, I referred to https://docs.khadas.com/vim3/HowToUseTengineSDK and converted the model to .tmfile (uint8). 

I have set the parameters as follows:
 **./quant_tool_uint8 -m ../convert_tool/yolov3.tmfile -i ~/data/git/npu/datesets/tengine_test_datasets_100/ -o yolov3_u8.tmfile -g 3,416,416 -a MINMAX  -w 0,0,0 -s 0.003922,0.003922,0.003922 -c 0 -t 4 -b 1 -y 416,416**

I am getting an error as shown below. Please help me solve this issue! 

**_khadas@Khadas:~/test/tengine_sdk/tengine_khadas_app/yolov3-picture/cv4_output$ ./tengine_khadas_yolov3_picture -m ../yolov3_u8.tmfile -i ..                 /holes1.png
tengine-lite library version: 1.4-dev
 failed
  ! mbedtls_net_connect returned -82

authorise failed ret[-82]
E [vsi_nn_QuantCheck:402]input_scale[-1334440575053054352202761503860850688.000000000000] * weight_scale[0.009132571518] != bias_scale[-121                 86873482064061837878380642435072.000000000000]
E [setup_node:456]Check node[2] CONV2D fail
Tengine Fatal: Pre-run subgraph(0) on TIMVX failed.
Tengine: Scheduler(sync) prerun failed.
Prerun graph failed**

Please note that the model is inferred on the PC and it is able to predict the objects perfectly. I am facing issues only when trying to run it using NPU.

Thank you!

BUG1989, 2021-06-30:
@AkkiSony 

Please give us the following file, and we check it.

- the loginfo when running the quant_tool_uint8;
- table_minmax.scale file;
- the origin fp32 tmfile.

AkkiSony, 2021-06-30:
@BUG1989 Thank you very much for replying. :)

1. Please find the loginfo screenshot attached.

[image]

2. table_minmax.scale file;
Please note that as I was not able to upload .scale format files, I just renamed the file to .txt
[table_minmax.scale.txt](https://github.com/OAID/Tengine/files/6738647/table_minmax.scale.txt)

3. the origin fp32 tmfile.
Can you please tell me which tmfile again? If it's the first generated tmfile(i think fp32), the size of that is 246MB. I am not able to upload that here.

BowShotDS, 2021-06-30:
[image]

Please change this to  -s 0.003922,0.003922,0.003922  and try again.

AkkiSony, 2021-06-30:
@BowShotDS Thank you for noticing the mistake I made. After flshing a new image with Gnome, I am getting the following output. I ma happy that now, it's able to predict the objects in my image. But I am still unsure if this is really using the NPU. Because, it takes a lot of time for inference.

Please find the attached screenshot below.

[image]

BUG1989, 2021-06-30:
The yolov3 inference time is 80ms with work on NPU.

AkkiSony, 2021-06-30:
@BUG1989 Thank you for your answer. I wanted an output as seen below. 
[image]
I think here, we have the  output of 5-TOPS.

But in my model, I am not getting any such output. So I am unsure if is working using NPU as my result is different compared to the official documentation. https://tengine-docs.readthedocs.io/en/latest/demo_guides/cuda.html

Please help me understand. Thank you very much! :)

AkkiSony, 2021-07-01:
@BUG1989 @BowShotDS 
As the inferernce time for Yolov3 is 80ms, I would like to use Mobilenet v1 SSD. While generating  the .tmfile, I got the following output. Can you please tell me where am I going wrong?

[image]