OAID/Tengine #786: tflite模型转换有不支持的算子结构，请问有什么方法吗？

luxingran, 2021-07-01:
---- Tengine Convert Tool ---- 

Version     : v1.0, 17:35:03 Jun 30 2021
Status      : float32
These 3ops  are not supported
{REDUCE_MAX,EXP,SUM,}
Create graph failed

mlbo, 2021-07-01:
提供一种解决思路可以尝试一下，可以先尝试[tflite转onnx](https://github.com/jackwish/tflite2onnx)，然后再使用[onnx-simplifier](https://github.com/daquexian/onnx-simplifier)，最后再[onnx转tmfile](https://convertmodel.com/)

qinhj, 2021-07-01:
我猜那几个不支持的算子，应该都是靠近最后output节点的后处理操作  直接编辑onnx砍掉吧~ 然后在demo里面去实现~

luxingran, 2021-07-02:
> 我猜那几个不支持的算子，应该都是靠近最后output节点的后处理操作 直接编辑onnx砍掉吧~ 然后在demo里面去实现~

 我输出模型的最后一层是进行Linear和softmax的concatenate操作，然后根据该output节点再进行后处理。现在可以确定是最后一层转换错误，请问转换onnx时是可以将已经训练好的模型进行砍掉最后一层的操作吗？

qinhj, 2021-07-04:
> > 我猜那几个不支持的算子，应该都是靠近最后output节点的后处理操作 直接编辑onnx砍掉吧~ 然后在demo里面去实现~
> 
> 我输出模型的最后一层是进行Linear和softmax的concatenate操作，然后根据该output节点再进行后处理。现在可以确定是最后一层转换错误，请问转换onnx时是可以将已经训练好的模型进行砍掉最后一层的操作吗？

转换时可以，直接根据你用的框架修改你需要转出的模型；转换后也可以，直接写python脚本，导入onnx包，加载onnx模型，然后对它进行编辑（比如直接砍掉、或是用其他算子代替等等）。[参考](https://github.com/OAID/Tengine/tree/tengine-lite/tools/optimize)