OAID/Tengine #798: whats faster coral TPU vs khadas NPU

taloot, 2021-07-04:
i have hard time googling and looking for information regarding this,
i used x86 with coral tpu, and i got 40ms on yolov5s512x512
and i have khadas in hand,, but i feel that khadas even if its 5Tops, and coral is 4Tops. but in real coral tpu faster.
this is my test on yolov5s...512x512 i want to see equivalent using  the am logic NPU or any another npu 
https://www.youtube.com/watch?v=D9IExho8pwo
i hope someone show me better results than this

BUG1989, 2021-07-05:
hi, thanks for your issue.

- what is the inference time of yolov5s at 512x512 with Khadas VIM3 NPU?
- some type op in yolov5s is not friendly for Khadas VIM3's NPI, we have a optimize network of yolov5s 640x640, it's cost 50ms with Khadas VIM3 NPU;
- If you have needs, we can help you design a more optimized network using Khadas VIM3 NPU.

taloot, 2021-07-05:
> Hi, @taloot thanks for your issue.
> 
> * what is the inference time of yolov5s at 512x512 with Khadas VIM3 NPU?
> * some type op in yolov5s is not friendly for Khadas VIM3's NPI, we have a optimize network of yolov5s 640x640, it's cost 50ms with Khadas VIM3 NPU;
> * If you have needs, we can help you design a more optimized network using Khadas VIM3 NPU.

 yes please i need to port the model i have for TPU, and compare it with coral tpu

taloot, 2021-07-11:
> Hi, @taloot thanks for your issue.
>
> * what is the inference time of yolov5s at 512x512 with Khadas VIM3 NPU?
> * some type op in yolov5s is not friendly for Khadas VIM3's NPI, we have a optimize [...]

can u show me a working video for this

AkkiSony, 2021-08-31:
@taloot Hi,
I see that you are working with VIM3 NPU and Coral TPU. Did you get to know which one is faster and better in terms of performance. I am working with the same AI accelerators.

Also, did you try working with other architectures other than yolov'x' ?
Thanks in advance! :)

taloot, 2021-08-31:
the vim3 didnt work for me at all
coral tpu works and every day it become better