OAID/Tengine #800: Create graph failed - MobileNet SSD v1

AkkiSony, 2021-07-05:
I succeeded in converting Yolov3 and utilizing the NPU using Tengine SDK. Now, I would like to use MobilNet SSD v1 on the NPU. 
During the conversion process, I got the following error. Please help me with the same. Thank you for your assistance.

[image]

AkkiSony, 2021-07-05:
Any suggestions please? @BUG1989

BUG1989, 2021-07-16:
Please use the Caffe SSD

AkkiSony, 2021-07-20:
I am using Caffe SSD pre trained model and want to check the inference time using NPU. During this process, I got the below problem. 
**_"Segmentation Fault"_**

Do you know how can I fix this problem?

[image]

AkkiSony, 2021-07-21:
@BUG1989 @debug202006 @kalcohol Can you please tell me which is the best possible way to get NPU working on MobileNet architecture. I am adamant on MobileNet, because it has good inference time compared to other models. 
Please suggest me that is compatible with Tengine model. Because, I will be using Tengine SDK to get it run on NPU.

AkkiSony, 2021-08-03:
Is there any idea to resolve this problem? I want to run "MobilNet SSD v1" on NPU. Either with NPU SDK or Tengine.

AkkiSony, 2021-08-09:
@BUG1989 @debug202006 @kalcohol I have a tensorflow mobilenet_v1_ssd model trained with my own custom dataset. Please can you tell me how can I successfully convert them to .tmfile and then utilize the NPU on VIM3?