OAID/Tengine #871: Tengine: Error when executing 'tm_yolov3_uint8.cpp'

AkkiSony, 2021-07-14:
Dear Tengine Team,
I have trained my model on yolov3 and converted it to .tmfile. Now, I want to do inference and run the model on Khadas VIM3 board. I came accross this file - https://github.com/OAID/Tengine/blob/tengine-lite/examples/tm_yolov3_uint8.cpp which looks like an inference code for yolov3. But when I execute this file, I am getting an error. 
Do I have to pass arguments like model (.tmfile), image etc while executing the .cpp code? 

Please have a look at my error output and suggest me where am I going wrong?

[image]

AkkiSony, 2021-07-15:
@BUG1989 @debug202006 @kalcohol Please, do you have any suggestion?

LosReturn, 2021-07-28:
Guys，I think u should link the opencv while using gcc 。
Well，there are something more meaningful hear。It’s time to learn Make or CMake that u can build the  yolov3-picture.

Give u an example to bind opencv：
```makefile
project(untitled1)

set(CMAKE_CXX_STANDARD 14)

find_package(OpenCV REQUIRED)

include_directories(${OpenCV_INCLUDE_DIRS})

add_executable(example example.cpp)
target_link_libraries(example ${OpenCV_LIBS})
```

Then，u can look at the  office link  to  resolve your question。
[CmakeList](https://github.com/OAID/Tengine/blob/tengine-lite/examples/CMakeLists.txt)
[README](https://github.com/OAID/Tengine/blob/tengine-lite/examples/README_EN.md)