OAID/Tengine #925: Segmentation fault on TIM VX

elejke, 2021-07-26:
Khadas VIM3
Ubuntu 20.04.2 LTS Linux 4.9.241
Galcore version 6.4.4.3.310723AAA

```
$ sudo find / -name "libOpenVX.so*"
/usr/lib/libOpenVX.so.1.3.0
```
Done `export LD_LIBRARY_PATH=/home/khadas/tengine_test/tengine-lite/3rdparty/tim-vx/lib/aarch64`:

```
$ echo $LD_LIBRARY_PATH
/home/khadas/tengine_test/tengine-lite/3rdparty/tim-vx/lib/aarch64
```

And have:

```
$ ./examples/tm_classification_timvx -m squeezenet_uint8.tmfile -i cat.jpg -r 1 -s 0.017,0.017,0.017 -r 10
Image height not specified, use default 227
Image width not specified, use default  227
Mean value not specified, use default   104.0, 116.7, 122.7
tengine-lite library version: 1.4-dev
```

BUG1989, 2021-07-27:
@elejke 
If you use the latest version of Khadas VIM3, you don't need to process the `3rdparty` implement, 
just only do the following 2 steps:

1. create the ` ./source/device/tim-vx/ ` directory
2. cmake enable ` -DTENGINE_ENABLE_TIM_VX=ON` option

please try it again, thanks.

elejke, 2021-07-27:
> @elejke
> If you use the latest version of Khadas VIM3, you don't need to process the `3rdparty` implement,
> just only do the following 2 steps:
> 
> 1. create the `./source/device/tim-vx/` directory
> 2. cmake enable ` -DTENGINE_ENABLE_TIM_VX=ON` option
> 
> please try it again, thanks.

```
Scanning dependencies of target tim_vx_internal
[ 27%] Building CXX object source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/tensor.cc.o
[ 27%] Building CXX object source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/graph.cc.o
[ 27%] Building CXX object source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/context.cc.o
[ 27%] Building CXX object source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/operation.cc.o
/home/khadas/tengine_test/tengine-lite/source/device/tim-vx/src/tim/vx/graph.cc:24:10: fatal error: tim/vx/graph.h: No such file or directory
   24 | #include "tim/vx/graph.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/build.make:76: source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/graph.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/khadas/tengine_test/tengine-lite/source/device/tim-vx/src/tim/vx/tensor.cc:24:10: fatal error: tim/vx/tensor.h: No such file or directory
   24 | #include "tim/vx/tensor.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/build.make:102: source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/tensor.cc.o] Error 1
/home/khadas/tengine_test/tengine-lite/source/device/tim-vx/src/tim/vx/context.cc:24:10: fatal error: tim/vx/context.h: No such file or directory
   24 | #include "tim/vx/context.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/build.make:63: source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/context.cc.o] Error 1
/home/khadas/tengine_test/tengine-lite/source/device/tim-vx/src/tim/vx/operation.cc:24:10: fatal error: tim/vx/operation.h: No such file or directory
   24 | #include "tim/vx/operation.h"
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/build.make:89: source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/src/tim/vx/operation.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:400: source/device/tim-vx/CMakeFiles/tim_vx_internal.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

BUG1989, 2021-07-27:
@elejke 

Can you understand this chinese guide ？
 https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual_zh.md

elejke, 2021-07-27:
> @elejke
> 
> Can you understand this chinese guide ？
> https://github.com/OAID/Tengine/blob/tengine-lite/doc/npu_tim-vx_user_manual_zh.md

I did the same as you said before and not created `./source/device/tim-vx/` directory but copied as it described in this manual. So, now build crashed after **tengine-lite** build done. On the **tm_benchmark**

```
[ 91%] Building C object source/CMakeFiles/tengine-lite.dir/utility/utils.c.o
[ 91%] Building C object source/CMakeFiles/tengine-lite.dir/utility/vector.c.o
[ 91%] Linking CXX shared library libtengine-lite.so
[ 91%] Built target tengine-lite
Scanning dependencies of target tm_benchmark
Scanning dependencies of target tm_hrnet_timvx
Scanning dependencies of target tm_yolov5s_timvx
Scanning dependencies of target tm_scrfd
[ 91%] Building CXX object benchmark/CMakeFiles/tm_benchmark.dir/common/timer.cc.o
[ 91%] Building CXX object examples/CMakeFiles/tm_scrfd.dir/tm_scrfd.cpp.o
[ 91%] Building CXX object examples/CMakeFiles/tm_yolov5s_timvx.dir/tm_yolov5s_timvx.cpp.o
[ 91%] Building CXX object examples/CMakeFiles/tm_hrnet_timvx.dir/tm_hrnet_timvx.cpp.o
[ 91%] Building CXX object benchmark/CMakeFiles/tm_benchmark.dir/tm_benchmark.cc.o
[ 91%] Building C object examples/CMakeFiles/tm_hrnet_timvx.dir/common/tengine_operations.c.o
[ 91%] Building C object examples/CMakeFiles/tm_yolov5s_timvx.dir/common/tengine_operations.c.o
[ 91%] Building C object examples/CMakeFiles/tm_scrfd.dir/common/tengine_operations.c.o
[ 91%] Linking CXX executable tm_benchmark
/usr/bin/ld: ../source/libtengine-lite.so: undefined reference to `vxTensorTableLookupLayer'
collect2: error: ld returned 1 exit status
make[2]: *** [benchmark/CMakeFiles/tm_benchmark.dir/build.make:102: benchmark/tm_benchmark] Error 1
make[1]: *** [CMakeFiles/Makefile2:456: benchmark/CMakeFiles/tm_benchmark.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Linking CXX executable tm_hrnet_timvx
[ 91%] Linking CXX executable tm_yolov5s_timvx
[ 91%] Linking CXX executable tm_scrfd
/usr/bin/ld: ../source/libtengine-lite.so: undefined reference to `vxTensorTableLookupLayer'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/tm_hrnet_timvx.dir/build.make:150: examples/tm_hrnet_timvx] Error 1
make[1]: *** [CMakeFiles/Makefile2:485: examples/CMakeFiles/tm_hrnet_timvx.dir/all] Error 2
/usr/bin/ld: ../source/libtengine-lite.so: undefined reference to `vxTensorTableLookupLayer'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/tm_yolov5s_timvx.dir/build.make:150: examples/tm_yolov5s_timvx] Error 1
make[1]: *** [CMakeFiles/Makefile2:514: examples/CMakeFiles/tm_yolov5s_timvx.dir/all] Error 2
/usr/bin/ld: ../source/libtengine-lite.so: undefined reference to `vxTensorTableLookupLayer'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/CMakeFiles/tm_scrfd.dir/build.make:150: examples/tm_scrfd] Error 1
make[1]: *** [CMakeFiles/Makefile2:543: examples/CMakeFiles/tm_scrfd.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

kalcohol, 2021-07-27:
Hi elejke, consider your VIM3 just flash a last ubuntu desktop image, and the system is fully updated(means galcore is the right version 6.4.4.3.310723AAA), then script below will help u to build a clean tengine.

before the very beginning, please remove builtin tengine libs(older version), use `sudo apt remove tengine-libs`; 
then the script is(should running on VIM3):
```bash
#!/bin/bash
if [ ! -d "Tengine" ]; then
    git clone https://github.com/OAID/Tengine.git
    git clone https://github.com/VeriSilicon/TIM-VX.git
    cp -r ./TIM-VX/include ./Tengine/source/device/timvx
    cp -r ./TIM-VX/src     ./Tengine/source/device/timvx
    cd Tengine && mkdir build && cd build
    cmake -DTENGINE_ENABLE_TIM_VX=ON -DTENGINE_BUILD_EXAMPLES=OFF ..
    make -j4 && make install
else
    cd Tengine/build && cmake -DTENGINE_ENABLE_TIM_VX=ON -DTENGINE_BUILD_EXAMPLES=OFF .. && make -j4 && make install
fi
```

from your building log, it seems that header files are not right.

please try the script again. holp it helps.

elejke, 2021-07-27:
> Hi elejke, consider your VIM3 just flashed a last ubuntu desktop image, and the system is fully updated(means galcore is the right version 6.4.4.3.310723AAA), then script below will help u to build a clean tengine.
> 
> before the very beginning, please remove builtin tengine libs(older version), use `sudo apt remove tengine-libs`;
> then the script is(should running on VIM3):
> 
> ```shell
> #!/bin/bash
> if [ ! -d "Tengine" ]; then
>     git clone https://github.com/OAID/Tengine.git
>     git clone https://github.com/VeriSilicon/TIM-VX.git
>     cp -r ./TIM-VX/include ./Tengine/source/device/tim-vx
>     cp -r ./TIM-VX/src     ./Tengine/source/device/tim-vx
>     cd Tengine && mkdir build && cd build
>     cmake -DTENGINE_ENABLE_TIM_VX=ON -DTENGINE_BUILD_EXAMPLES=OFF ..
>     make -j4 && make install
> else
>     cd Tengine/build && cmake -DTENGINE_ENABLE_TIM_VX=ON -DTENGINE_BUILD_EXAMPLES=OFF .. && make -j4 && make install
> fi
> ```
> 
> from your building log, it seems that header files are not right.
> 
> please try the script again. holp it helps.

```
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/lock.c.o
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/log.c.o
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/math.c.o
/home/khadas/src/clean_build/Tengine/source/utility/float.c: In function ‘pxr24_to_fp32’:
/home/khadas/src/clean_build/Tengine/source/utility/float.c:198:8: note: parameter passing for argument of type ‘pxr24_pack_t’ {aka ‘struct pxr24_pack’} changed in GCC 9.1
  198 | fp32_t pxr24_to_fp32(pxr24_pack_t package)
      |        ^~~~~~~~~~~~~
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/mem_stat.c.o
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/sys_port.c.o
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/utils.c.o
[100%] Building C object source/CMakeFiles/tengine-lite-static.dir/utility/vector.c.o
[100%] Linking CXX static library libtengine-lite-static.a
[100%] Built target tengine-lite-static
make: *** [Makefile:141: all] Error 2
```

I will try to flash new clean ubuntu...

closed 2021-07-28 by elejke