OAID/Tengine #1028: 输入图片尺寸高h和宽w不相等时报错Segmentation Fault

Yanzuo1019, 2021-08-14:
pytorch导出预训练resnet50模型，去掉fc层，输入设置为256x128，经过量化成uint8模型以后，发现在tm_classification_timvx基础上小修改后，-g 256,128时会出现Segmentation Fault，但是-g 256,256或者-g 128,128 却是能正常运行的，明明模型的输入设置是256x128，为什么经过resize的图片大小和模型设置不一致时能正常输出结果，一致的时候却会报错呢，下面是我的测试代码以及onnx和uint8模型，目前能够确认是在**run_graph**函数这里出错了

`/*
 * Licensed to the Apache Software Foundation (ASF) under one
 * or more contributor license agreements.  See the NOTICE file
 * distributed with this work for additional information
 * regarding copyright ownership.  The ASF licenses this file
 * to you under the Apache License, Version 2.0 (the
 * License); you may not use this file except in compliance
 * with the License.  You may obtain a copy of the License at
 *
 *   http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing,
 * software distributed under the License is distributed on an
 * AS IS BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY
 * KIND, either express or implied.  See the License for the
 * specific language governing permissions and limitations
 * under the License.
 */

/*
 * Copyright (c) 2020, OPEN AI LAB
 * Author: qtang@openailab.com
 */

#include <stdlib.h>
#include <stdio.h>

#include "common.h"
#include "tengine/c_api.h"
#include "tengine_operations.h"

#define DEFAULT_IMG_H        256
#define DEFAULT_IMG_W        128
#define DEFAULT_SCALE1       0.0174
#define DEFAULT_SCALE2       0.0175
#define DEFAULT_SCALE3       0.0171
#define DEFAULT_MEAN1        103.53
#define DEFAULT_MEAN2        116.28
#define DEFAULT_MEAN3        123.675
#define DEFAULT_LOOP_COUNT   1
#define DEFAULT_THREAD_COUNT 1
#define DEFAULT_CPU_AFFINITY 255

void get_input_uint8_data(const char* image_file, uint8_t* input_data, int img_h, int img_w, float* mean, float* scale,
                          float input_scale, int zero_point)
{
    image img = imread_process(image_file, img_w, img_h, mean, scale);

    float* image_data = (float*)img.data;
    for (int i = 0; i < img_w * img_h * 3; i++)
    {
        int udata = round(image_data[i] / input_scale + (float)zero_point);
        if (udata > 255)
            udata = 255;
        else if (udata < 0)
            udata = 0;

        input_data[i] = udata;
    }

    free_image(img);
}

int tengine_reid(const char* model_file, const char* image_file, int img_h, int img_w, float* mean, float* scale,
                     int loop_count, int num_thread, int affinity)
{
    /* set runtime options */
    struct options opt;
[... 219 lines cut ...]
    {
        scale[0] = DEFAULT_SCALE1;
        scale[1] = DEFAULT_SCALE2;
        scale[2] = DEFAULT_SCALE3;
        fprintf(stderr, "Scale value not specified, use default  %.4f, %.4f, %.4f\n", scale[0], scale[1], scale[2]);
    }

    if (mean[0] == -1.0 || mean[1] == -1.0 || mean[2] == -1.0)
    {
        mean[0] = DEFAULT_MEAN1;
        mean[1] = DEFAULT_MEAN2;
        mean[2] = DEFAULT_MEAN3;
        fprintf(stderr, "Mean value not specified, use default   %.1f, %.1f, %.1f\n", mean[0], mean[1], mean[2]);
    }

    if (tengine_reid(model_file, image_file, img_h, img_w, mean, scale, loop_count, num_thread, cpu_affinity) < 0)
        return -1;

    return 0;
}
`

链接：https://pan.baidu.com/s/1kpgkHlWuMq0yoZZCK3C92A 
提取码：m5zc 
--来自百度网盘超级会员V4的分享

LJoson, 2021-08-17:
我看了一下，对应示例，我个人理解，原因可能是默认尺寸是相等的，对应的tensor以及构建gragh,是根据尺寸想等的情况下处理，所以不相等时可能出现异常，而且项目没有对这个异常进行处理（依稀记得虫叔那篇文章提过输入尺寸的规则），更具体的好像要研究一下对应参数，以及run gragh流程了

Yanzuo1019, 2021-08-19:
是的，我尝试了一下khadas自家提供的sdk，它们同样调用的是TIM-VX，没有发现这个问题，应该是Tengine这边出bug了

> 我看了一下，对应示例，我个人理解，原因可能是默认尺寸是相等的，对应的tensor以及构建gragh,是根据尺寸想等的情况下处理，所以不相等时可能出现异常，而且项目没有对这个异常进行处理（依稀记得虫叔那篇文章提过输入尺寸的规则），更具体的好像要研究一下对应参数，以及run gragh流程了