OAID/Tengine #1038: 关于int8 pooling 和 uint8 pooling实现，为什么uint8需要转成float32呢，这样子是不是浪费空间和时间啊？

gftd, 2021-08-16:
关于int8 pooling 和 uint8 pooling实现，为什么uint8需要转成float32呢，这样子是不是浪费空间和时间啊？
具体文件是
pooling_kernel_ref_int8.c

    int8_t* input_int8 = (int8_t*)input_tensor->data;
    int8_t* output_int8 = (int8_t*)output_tensor->data;

pooling_kernel_ref_uint8.c

    float* input_fp32 = (float*)sys_malloc(input_tensor->elem_num * sizeof(float));
    float* output_fp32 = (float*)sys_malloc(output_tensor->elem_num * sizeof(float));
    for (int i = 0; i < input_tensor->elem_num; i++)
        input_fp32[i] = (input_uint8[i] - input_zero) * input_scale;
    float* input = input_fp32;
    float* output = output_fp32;

LJoson, 2021-08-17:
可以从int8，uint8,float数据类型理解看看，我们如果是int数据，那么范围是-1到1（举例子），那么float也是-1到1，而uint是0到1，那么，如果不进行数据转换，精度至少会掉一半以上，所以这里uint到float更多可以从数据归一化，类型统一这边理解，至于空间与时间的浪费，或许时间可能是多一点，但是也不会多久，空间也是，CPU的isa只认识数据，后面的处理几乎都是对有符号数据处理，在能保证精度的情况下，多点时间与空间开支是可行的，不然就是车毁人完

LJoson, 2021-08-17:
https://blog.csdn.net/weixin_37707670/article/details/111055256?utm_medium=distribute.pc_relevant.none-task-blog-2%7Edefault%7EBlogCommendFromMachineLearnPai2%7Edefault-9.control&depth_1-utm_source=distribute.pc_relevant.none-task-blog-2%7Edefault%7EBlogCommendFromMachineLearnPai2%7Edefault-9.control

gftd, 2021-08-18:
> 可以从int8，uint8,float数据类型理解看看，我们如果是int数据，那么范围是-1到1（举例子），那么float也是-1到1，而uint是0到1，那么，如果不进行数据转换，精度至少会掉一半以上，所以这里uint到float更多可以从数据归一化，类型统一这边理解，至于空间与时间的浪费，或许时间可能是多一点，但是也不会多久，空间也是，CPU的isa只认识数据，后面的处理几乎都是对有符号数据处理，在能保证精度的情况下，多点时间与空间开支是可行的，不然就是车毁人完

那么为啥int8时候不转成float呢，int8和uint8表达的精度差不多啊？

LeiWang1999, 2021-08-19:
@gftd 我记得是int8的MaxPooling没有做，因为Maxpooling有量化透传机制，input_scale和output_scale是一样的，这样就可以直接省略不写啦，avgpooling还是有的（