OAID/Tengine #1039: 关于使用Arm64 OHOS 编译脚本的问题

LJoson, 2021-08-16:
编译脚本中以下部分
`mkdir build-ohos-armeabi-v7a
pushd build-ohos-armeabi-v7a
%CMAKE_EXE% -G Ninja -DCMAKE_TOOLCHAIN_FILE="%OHOS_NDK%/build/cmake/ohos.toolchain.cmake"  -DCMAKE_MAKE_PROGRAM=%NINJA_EXE%  -DOHOS_ARCH="armeabi-v7a" -DCMAKE_BUILD_WITH_INSTALL_RPATH=ON .. 
%CMAKE_EXE% --build . --parallel %NUMBER_OF_PROCESSORS%
%CMAKE_EXE% --build . --target install
popd`
编译过程出现了不少类似以下报错导致无法编译成功
`../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:725:2: error: instruction requires: VFP2
 vmla.f32 s24, s2, s12
 ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:726:2: error: instruction requires: VFP2
 vmla.f32 s25, s3, s12
 ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:727:2: error: instruction requires: VFP2
 vmla.f32 s26, s4, s12
 ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:728:2: error: instruction requires: VFP2
 vmla.f32 s27, s5, s12
 ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:730:9: error: invalid instruction
        vadd.f32 d8, d8, d16
        ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:731:9: error: invalid instruction
        vadd.f32 d9, d9, d16
        ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:732:9: error: invalid instruction
        vadd.f32 d12, d12,d16
        ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:733:9: error: invalid instruction
        vadd.f32 d13, d13,d16
        ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:735:2: error: instruction requires: NEON
 vmax.f32 q4, q4, q9
 ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:736:2: error: instruction requires: NEON
 vmax.f32 q6, q6, q9
 ^
../../source/device/cpu/op/conv/cortex-a/armv7/./dw_k3s1p1.S:738:2: er`
是啥原因捏

LJoson, 2021-08-16:
[image]
再补一个错误信息图

cmdbug, 2021-08-17:
> [image]
> 再补一个错误信息图

这个不是提示了没有找到 opencv 了嘛，可以加上 opencv 的参数，或直接在 examplex/CMakeLists.txt 103行左右设置 opencv 的安装目录。

LJoson, 2021-08-17:
> > [image]
> > 再补一个错误信息图
> 
> 这个不是提示了没有找到 opencv 了嘛，可以加上 opencv 的参数，或直接在 examplex/CMakeLists.txt 103行左右设置 opencv 的安装目录。(如: D:/OpenCV/build/x64/vc14/lib 目录下要有 OpenCVConfig.cmake 这个文件)

不是opencv引起的，我没有使用opencv，其对应编译的example，tengine是没有编译的，图里主要是想体现一下所用的clang等工具信息