OAID/Tengine #1043: 关于convert tool的编译问题

huoshuai-dot, 2021-08-17:
1、编译要求protobuf的版本》3.10.0 但是 想知道官方是怎么使用apt-get install命令安装高版本的protobuf的？
2、自己编译安装好高版本protobuf之后 不修改cmakelist的话 没办法通过find_package()找到对应的protobuf，这个该怎么配置？
3、强行设置，命令如下
set(Protobuf_INCLUDE_DIR "/usr/local/protobuf/include")
set(Protobuf_LIBRARIES "/usr/local/protobuf/lib")
set(Protobuf_PROTOC_EXECUTABLE "/usr/local/protobuf/bin/protoc")
报如下错误
tools/convert_tool/CMakeFiles/convert_tool.dir/onnx/onnx.pb.cc.o:(.data.rel+0x178): more undefined references to `google::protobuf::internal::fixed_address_empty_string[abi:cxx11]' follow
tools/convert_tool/CMakeFiles/convert_tool.dir/caffe/caffe2tengine.cpp.o: In function `caffe_serializer::load_graph_node(graph*, te_caffe::NetParameter, te_caffe::NetParameter)':
caffe2tengine.cpp:(.text+0x7f27): undefined reference to `google::protobuf::RepeatedPtrField<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::size() const'
caffe2tengine.cpp:(.text+0x7f39): undefined reference to `google::protobuf::RepeatedPtrField<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::Get(int) const'
caffe2tengine.cpp:(.text+0x8009): undefined reference to `google::protobuf::RepeatedPtrField<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::size() const'
caffe2tengine.cpp:(.text+0x801b): undefined reference to `google::protobuf::RepeatedPtrField<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::Get(int) const'
tools/convert_tool/CMakeFiles/convert_tool.dir/caffe/caffe2tengine.cpp.o: In function `caffe_serializer::load_text_file(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, te_caffe::NetParameter&)':
caffe2tengine.cpp:(.text+0x94ca): undefined reference to `google::protobuf::io::IstreamInputStream::IstreamInputStream(std::istream*, int)'
这个问题该怎么解决？
希望各位大佬帮忙解答一二

LJoson, 2021-08-17:
可以提供当前环境，以及对应您编译安装的protobuf的一些相关信息嘛，

huoshuai-dot, 2021-08-18:
> 可以提供当前环境，以及对应您编译安装的protobuf的一些相关信息嘛，

我用的是wsl 18.04 protobuf version libprotoc 3.17.3, cmake version 3.21.0-rc2

huoshuai-dot, 2021-08-18:
set(Protobuf_LIBRARIES "/usr/local/protobuf/lib")这个没有set对 
改成set(Protobuf_LIBRARIES "/usr/local/protobuf/lib/libprotobuf.a")就ok了

closed 2021-08-18 by huoshuai-dot