OAID/Tengine #1046: 关于onnx模型转换为tengine后float32都无法对齐

unsky, 2021-08-18:
自己随便造了一层卷积层产生了onnx,该实验主要看一个卷积有没有对齐:
代码:
```python
import cv2
import numpy as np
from tengine import tg

import onnxruntime as ort

ort_session = ort.InferenceSession('conv.onnx')
in_tensor = np.random.randn(1, 3, 224, 224).astype(np.float32)
outputs = ort_session.run(None, {'input': in_tensor})
print("onnx out:", outputs[0].mean(), outputs[0].max(), outputs[0].min() )
tm_file = "conv.tmfile"
graph = tg.Graph(None, 'tengine', tm_file

input_tensor = graph.getTensorByName("input")
dims = [1, 3, 224, 224]
input_tensor.shape = dims
input_tensor.buf = in_tensor
graph.preRun()
graph.run(1) # 1 is blocking
out = graph.getTensorByName("output")
out = np.array(out.buf)
out = out.reshape((1,4,224,224))
print ("tengine:", out.mean(), out.max(), out.min())
```

  `
                               mean                max                 min

onnx out: 0.02329052 2.6034281 -2.7885275

tengine: 0.035146895542775565 2.739078998565674 -2.6004815101623535
`

一个卷积的差别就这么大了?

xiguadong, 2021-08-19:
请问你用的是最新的代码吗？ 能否提供下onnx和tmfile模型呢？

xiguadong, 2021-08-19:
你好，请更新到最新版本的Tengine，重新python setup.py install。

xiguadong, 2021-08-19:
你好，请更新到最新版本再试一下，我刚刚在本地试过。结果如下：
onnx out: -0.24560605 0.34762448 -0.9356478
self.dtype.enum: 0
tengine: -0.24560605816222217 0.34762439131736755 -0.9356477856636047

pytengine推理之前是有bug的，目前已经修复。这个pr有说明原因：https://github.com/OAID/Tengine/pull/1032
另外tegine提供了和onnx的对齐工具，可以尝试使用 https://github.com/OAID/Tengine/tree/tengine-lite/tools/align_tool

unsky, 2021-08-19:
ok ths

closed 2021-08-19 by unsky