OAID/Tengine #1092: onnx convert tool 和quan

unsky, 2021-08-29:


unsky, 2021-08-29:
https://github.com/OAID/Tengine/blob/5c50666cd3de149b5b79d3974d3b630a333c5ceb/tools/convert_tool/onnx/onnx2tengine.cpp#L1232

unsky, 2021-08-29:
```
static int load_interp(ir_graph_t* graph, ir_node_t* node, const onnx::NodeProto& onnx_node)
{
    std::string mode = "nearest";
    for (int k = 0; k < onnx_node.attribute_size(); k++)
    {
        const onnx::AttributeProto& attr = onnx_node.attribute(k);

        if (attr.name() == "mode")
        {
            mode = attr.s();
        }
    }
    if (mode != "nearest")
    {
        struct interp_param* interp_param = (struct interp_param*)node->op.param_mem;

        if (onnx_node.input_size() == 1)
        {
            for (int k = 0; k < onnx_node.attribute_size(); k++)
            {
                const onnx::AttributeProto& attr = onnx_node.attribute(k);
                if (attr.name() == "scales")
                {
                    if (attr.floats_size() == 4)
                    {
                        float num0 = attr.floats(0);
                        float num1 = attr.floats(1);
                        float num2 = attr.floats(2);
                        float num3 = attr.floats(3);
                        interp_param->height_scale = num2 / num0;
                        interp_param->width_scale = num3 / num1;
                    }
                    else
                    {
                        interp_param->height_scale = attr.f();
                        interp_param->width_scale = attr.f();
                    }
                }
            }
        }
        else
        {
            const std::string& input_name = onnx_node.input(1);
            ir_tensor_t* tensor = find_tensor(graph, input_name);
            float* data = (float*)tensor->data;

            interp_param->height_scale = data[2];
            interp_param->width_scale = data[3];
        }
        if (mode == "nearest")
        {
            interp_param->resize_type = 1;
        }
        else if (mode == "bilinear" || mode == "linear")
        {
            interp_param->resize_type = 2;
        }
    }
    else
    {
        if (change_node_op(node, OP_RESIZE) < 0)
        {
            return -1;
        }
        struct resize_param* resize_param = (struct resize_param*)node->op.param_mem;

        if (onnx_node.input_size() == 2)
        {
            const std::string& input_name = onnx_node.input(1);
            ir_tensor_t* tensor = find_tensor(graph, input_name);
            float* data = (float*)tensor->data;
            resize_param->scale_h = data[2];
            resize_param->scale_w = data[3];
        }
        else
        {
            resize_param->scale_w = 1.f;
            resize_param->scale_h = 1.f;
        }
    }

    return 0;
}
```
  if (mode != "nearest") 应该shi ===?

unsky, 2021-08-29:
另外:
```c++
            const std::string& input_name = onnx_node.input(1);
            ir_tensor_t* tensor = find_tensor(graph, input_name);
```
返回的zhizhen不对

BUG1989, 2021-08-31:
@FeiGeChuanShu Please check out it, thanks.

FeiGeChuanShu, 2021-08-31:
@unsky 当导出onnx时opset=9 or 10的时候会导出这个op，如果是nearest，resize和interp的结果是一样的（这里没有问题），如果是bilinear，导出onnx时opset=11 or higher 结果也没问题，因为9 or 10的upsample bilinear结果跟pytorch也不能对齐。因此建议导出onnx时opset设置11 or higher。以下是onnx官方答复：
https://github.com/onnx/onnx/issues/2535

closed 2021-09-01 by BUG1989