OAID/Tengine #1098: Cannot convert grouped convolution in ONNX?

Goose-Bomb, 2021-08-31:
NN structure:
[image]

Log from convert_tool:
```log
---- Tengine Convert Tool ---- 

Version     : v1.0, 14:33:02 Aug 31 2021
Status      : float32

----------onnx2tengine begin----------
Model op set is :13
Error: input elem num(148225) != reshaped elem num(4743200)
Tengine FATAL: Infer node(id: 12, op: Reshape) shape failed.
Convert model failed.
```

ONNX file:
[ncc.onnx.tar.gz](https://github.com/OAID/Tengine/files/7081955/ncc.onnx.tar.gz)

The Conv node is a grouped convolution with `groups == 32`, and the input tensor have 32 channels, so the output feature map's shape should be `1 x 32 x 385 x 385`. However, it seemed that the output tensor produced by tengine is actually in `1 x 1 x 385 x 385` == 148255 elements, so the subsequent Reshape Op failed because of elements number mismatch.

cccxinli, 2021-09-01:
Hello, your model cannot be opened, could you check and re-upload it?

cccxinli, 2021-09-02:
May fixed. Pull code and try again.

Goose-Bomb, 2021-09-02:
The latest version convert tool is able to convert the model, but the feature map shape after convolution is still incorrect as I mentioned before. Seems that the groups attribute is ignored by the Conv node.

Converted tmfile structure:
[image]

Model files: (original onnx and converted tmfile)
[ncc.tar.gz](https://github.com/OAID/Tengine/files/7096195/ncc.tar.gz)

cccxinli, 2021-09-02:
You are right. I just found the weight of this conv, is a variable tensor rather than a constant tensor.

Goose-Bomb, 2021-09-02:
> You are right. I just found the weight of this conv, is a variable tensor rather than a constant tensor.

Yes, what this NN do is to compute the normalized cross correlation between two feature maps, so the weights of the Conv is reshaped from an input tensor.