OAID/Tengine #1108: onnx model for tm_scrfd.cpp

rahulsharma11, 2021-09-01:
Hi,
in tm_scrfd.cpp, there is given that onnx model is from insightface scrfd.

Question-
Is there any updation done while model conversion to tmfile?
Or can you provide the onnx model for scrfd_2.5g_kps.tmfile/scrfd_10g_kps.tmfile ?

I tried to use Tengine-convert tool for the same but no results are there.

Thanks.

FeiGeChuanShu, 2021-09-02:
[image]
if you want use your own scrfd model ,you maybe should change the export code of scrfd like this.

FeiGeChuanShu, 2021-09-02:
[image]
hi, maybe you should change the export code like this.

rahulsharma11, 2021-09-02:
Hi, Thanks for the reply.

I did that already but no result. Throwing error - "get graph tensor failed".

Also, i checked with Netron, there are no shapes in output layer.

rahulsharma11, 2021-09-03:
is there possiblity if i can get onnx (scrfd_10g_kps.onnx) model which can be converted to tmfile to be used in ncnn ?