OAID/Tengine #1125: Build Tengine with TensorRT failed in Jetson nano!

ghimiredhikura, 2021-09-07:
Enabling tensorrt in jetson-nano failed with the following error. 

```ruby
looser throw specifier for ‘virtual void Logger::log(nvinfer1::ILogger::Severity, const char*)’
     void log(Severity severity, const char* msg) override
```

- Compile log: 
```ruby
Tengine/build-linux-trt$ cmake -DTENGINE_ENABLE_TENSORRT=ON -DTENGINE_BUILD_CPP_API=OFF -DTENGINE_ONLINE_REPORT=OFF ..                                                                                       -- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found CUDAToolkit: /usr/local/cuda/include (found version "10.2.300")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CUDNN: /usr/include (found version "8.2.1")
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Found OpenMP_C: -fopenmp (found version "4.5")
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Found OpenCV: /usr (found version "4.1.1")
--
--
-- Infomation Summary:
--
-- CMake infomation:
--   - CMake version:              3.21.2
--   - CMake generator:            Unix Makefiles
--   - CMake building tools:       /usr/bin/make
--   - Target System:              Linux
--   - Target CPU arch:            ARM
--   - Target building type:       release
--   - Target CPU bus width:       64 Bit
--
-- tengine-lite toolchain infomation:
--   Cross compiling: FALSE
--   C/C++ compilier:
--     - C   standard version:     C99
--     - C   standard required:    TRUE
--     - C   standard extensions:  ON
--     - C   compilier version:    7.5.0
--     - C   compilier:            /usr/bin/cc
--     - C++ standard version:     C++11
--     - C++ standard required:    TRUE
--     - C++ standard extensions:  OFF
--     - C++ compilier version:    7.5.0
--     - C++ compilier:            /usr/bin/c++
--   C/C++ compilier flags:
--     - C   compilier flags:
--     - C++ compilier flags:
--   OpenMP:
[... 191 lines cut ...]
/usr/include/aarch64-linux-gnu/NvInfer.h:7929:25: note: declared here
     TRT_DEPRECATED void destroy() noexcept
                         ^~~~~~~
/home/ghimire/Desktop/Tengine/source/device/tensorrt/trt_executor.cc: In member function ‘void TensorRTEngine::SetOption(trt_opt_t*)’:
/home/ghimire/Desktop/Tengine/source/device/tensorrt/trt_executor.cc:859:21: warning: ‘void nvinfer1::IBuilder::destroy()’ is deprecated [-Wdeprecated-declarations]
         bd->destroy();
                     ^
In file included from /home/ghimire/Desktop/Tengine/source/device/tensorrt/trt_executor.hpp:40:0,
                 from /home/ghimire/Desktop/Tengine/source/device/tensorrt/trt_executor.cc:25:
/usr/include/aarch64-linux-gnu/NvInfer.h:7929:25: note: declared here
     TRT_DEPRECATED void destroy() noexcept
                         ^~~~~~~
[ 38%] Building CXX object source/CMakeFiles/tengine-lite-static.dir/device/tensorrt/op/trt_batchnorm.cc.o
source/CMakeFiles/tengine-lite-static.dir/build.make:2849: recipe for target 'source/CMakeFiles/tengine-lite-static.dir/device/tensorrt/trt_executor.cc.o' failed
make[2]: *** [source/CMakeFiles/tengine-lite-static.dir/device/tensorrt/trt_executor.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:355: recipe for target 'source/CMakeFiles/tengine-lite-static.dir/all' failed
make[1]: *** [source/CMakeFiles/tengine-lite-static.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```

Any idea, please? 

Deepak

leeyunhome, 2022-02-22:
Hello,

You have resolved this issue?

Thank you.

ringwraith, 2022-09-14:
+++ b/source/device/tensorrt/trt_helper.hpp
@@ -100,7 +100,7 @@ public:
     Logger(nvinfer1::ILogger::Severity severity = nvinfer1::ILogger::Severity::kINFO)
         : severity_(severity){};

-    void log(Severity severity, const char* msg) override
+    void log(Severity severity, const char* msg) noexcept override