OAID/Tengine #1131: 如何从内存中读取模型

joshua19881228, 2021-09-12:
请问tengine-lite是否支持从内存中读取模型。在文档内没有看到相关的说明，但源码中看到有相关的处理，没有看太明白是如何处理的

MeredithLeaf, 2021-09-13:
你好，请问在“源码中看到有相关的处理” 这段源码的路径是在哪里？

MeredithLeaf, 2021-09-13:
> 请问tengine-lite是否支持从内存中读取模型。在文档内没有看到相关的说明，但源码中看到有相关的处理，没有看太明白是如何处理的

你好，请问在“源码中看到有相关的处理” 这段源码的路径是在哪里？

MeredithLeaf, 2021-09-13:
> 请问tengine-lite是否支持从内存中读取模型。在文档内没有看到相关的说明，但源码中看到有相关的处理，没有看太明白是如何处理的

graph_t create_graph(context_t context, const char* model_format, const char* file_name, ...)
看这个接口的实现，支持使用memory方式加载模型，model_format需要填貌似tengine:m这样的方式，可以试下。

joshua19881228, 2021-09-14:
> > 请问tengine-lite是否支持从内存中读取模型。在文档内没有看到相关的说明，但源码中看到有相关的处理，没有看太明白是如何处理的
> 
> graph_t create_graph(context_t context, const char* model_format, const char* file_name, ...)
> 看这个接口的实现，支持使用memory方式加载模型，model_format需要填貌似tengine:m这样的方式，可以试下。

我在看代码的时候也猜测是tengine:m这样的方式来做，不过tengine提示没有这个模块。不知道是不是在初始化的时候需要有特殊的操作

[image]

xiguadong, 2021-09-17:
这个是一个bug，待修复 @joshua19881228 @MeredithLeaf

joshua19881228, 2021-09-18:
> 这个是一个bug，待修复 @joshua19881228 @MeredithLeaf

好的，谢谢

closed 2021-09-18 by joshua19881228

wrchen-voxel, 2021-09-27:
> 我在看代码的时候也猜测是tengine:m这样的方式来做，不过tengine提示没有这个模块。不知道是不是在初始化的时候需要有特殊的操作
>
> [image]

create_graph 把find_serializer_via_name的model_format试着做下替换似乎就可以跑起来

```
graph_t create_graph(context_t context, const char* model_format, const char* file_name, ...)
{
...
        int ret = 0;

        const char *p, *q;
        unsigned char tmp;
        p = strchr(model_format, ':');
        tmp = (unsigned char)(strchr(model_format, ':')-model_format);
        q = (tmp > 0)?strndup(model_format, tmp):strdup(model_format);

        struct serializer* loader = find_serializer_via_name(q);
        if (loader == NULL)
        {
            TLOG_ERR("Tengine: No matched serializer(name: %s) found.\n", q);
            return create_graph_error;
        }

        va_list ap;
        va_start(ap, file_name);

        // const char* p = strchr(model_format, ':');

        // load from file or memory
        if (NULL == p)
...
}
```

但对应这块存储模型的内存 在跑完 destroy_graph  时 unload_graph 做sys_free回收时 好像也要重新处理