OAID/Tengine #1146: ONNX2TMFILE工具异常问题

GuideWsp, 2021-09-16:
今天在用预编译版的转换工具对EDSR超分模型进行转换时 ，突然发现：
在EDSR的上采样部分转换成ONNX后为DepthToSpace，而经过转换工具转换为tmfile后就变成了Space'T'oDepth
[image]
烦请大佬出来给个解决方案哈

FeiGeChuanShu, 2021-09-16:
@GuideWsp 方便发一下onnx文件吗

GuideWsp, 2021-09-16:
我用的转换工具、两个模型的onnx以及convert.sh都已经上传百度网盘，麻烦帮忙看一下，如能加一下微信(AIWalker-Happy)就更好了，谢谢。
链接：https://pan.baidu.com/s/1epGn-J_LNHt1dVP37FCYZw 
提取码：aiha

FeiGeChuanShu, 2021-09-16:
@GuideWsp 
[image]
试过了转换没问题，应该是你用的工具太老了，建议使用下面地址中的convert tool
https://github.com/OAID/Tengine/tree/tengine-lite/tools/convert_tool

GuideWsp, 2021-09-16:
转换工具用的主页提供的链接，因为不想去编译所以直接下载的线程的。看来主页的链接该更新一下了
[image]

GuideWsp, 2021-09-16:
@FeiGeChuanShu 谢谢。看来偷懒是不行的，还是要重新编译一下才行。

GuideWsp, 2021-09-17:
> @GuideWsp
> [image]
> 试过了转换没问题，应该是你用的工具太老了，建议使用下面地址中的convert tool
> https://github.com/OAID/Tengine/tree/tengine-lite/tools/convert_tool

重新编译了转换工具，EDSR这个模型转换OK，但是ESRGAN模型转换有问题。麻烦大佬试一下您那边能否转换成功,谢谢
[image]

FeiGeChuanShu, 2021-09-18:
@GuideWsp 经验证，ESRGAN模型请先使用以下工具onnx-sim将你的模型优化一下，然后就可以转换成功了，部分node input的解析问题。
https://github.com/daquexian/onnx-simplifier

GuideWsp, 2021-09-22:
@FeiGeChuanShu 谢谢帮忙测试。

closed 2021-09-22 by GuideWsp