OAID/Tengine #1147: 有不支持的层噢，能写plugin吗

zhangchenchen-novauto, 2021-09-16:
```
These 2op are not supported
{Sqrt,Reciprocal,}
Create graph failed
```

xiguadong, 2021-09-17:
请附上原始onnx模型

zhangchenchen-novauto, 2021-09-17:
@xiguadong 
[AnimeGANv2slim.onnx.txt](https://github.com/OAID/Tengine/files/7182050/AnimeGANv2slim.onnx.txt)

把后缀改一下就好了呢

xiguadong, 2021-09-22:
@zhangchenchen-novauto 你好，已经提交了commit . 

https://github.com/OAID/Tengine/pull/1159，通过后可以pull下最新的试试。不过根据对齐结果所示，upsample op有所不同，可能导致精度有损。
[compare.txt](https://github.com/OAID/Tengine/files/7210276/compare.txt) 这里是我拿你的模型和转出的tmfile做的一个对齐结果。
对齐工具在https://github.com/OAID/Tengine/tree/tengine-lite/tools/align_tool这里，如果要运行该脚本，建议放到服务器上运行，非常吃内存。

这个op后续会进行修复，中秋摸鱼太狠了，修复有点慢..

zhangchenchen-novauto, 2021-09-23:
@xiguadong 
哇 辛苦啦 辛苦啦

CallmeZhangChenchen, 2021-09-30:
@xiguadong 

我好像用最新的代码没有跑通，是不是还没有pull  
[AnimeGANv2slim.tmfile.txt](https://github.com/OAID/Tengine/files/7258805/AnimeGANv2slim.tmfile.txt)

```
Image height not specified, use default 384
Image width not specified, use default  384
Scale value not specified, use default  127.5, 127.5, 127.5
Mean value not specified, use default   127.5, 127.5, 127.5
tengine-lite library version: 1.5-dev
CPU: failed to run node 17, generator/G_MODEL/A/LayerNorm/batchnorm/mul_1
run subgraph 0 error!
Run graph failed
```

xiguadong, 2021-10-03:
是用最新的转换工具转的吗

CallmeZhangChenchen, 2021-10-08:
@xiguadong 
是最新的

[image]