OAID/Tengine #1176: 在执行tools/optimize/yolov5s-opt.py时报错

BoXingJinQiu, 2021-10-11:
在对yolov5转的onnx模型进行优化时报错，请问是不是预训练模型更新了，优化的代码没有更新？

---- Tengine YOLOv5 Optimize Tool ----

Input model      : yolov5s.v4.onnx
Output model     : yolov5s.v4.opt.onnx
Input tensor     : 167
Output tensor    : 381,420,459
[Quant Tools Info]: Step 0, load original onnx model from yolov5s.v4.onnx.
256
[Quant Tools Info]: Step 1, Remove the focus and postprocess nodes.
Traceback (most recent call last):
  File "yolov5s-opt.py", line 222, in <module>
    main()
  File "yolov5s-opt.py", line 184, in main
    new_nodes = cut_focus_output(old_node, in_tensor, out_tensor)
  File "yolov5s-opt.py", line 82, in cut_focus_output
    output_pass[node_dict[out_name[i]]] = 2
KeyError: '381'

xukefang, 2021-11-18:
+1

cmd:python3 yolov5s-opt.py --input yolov5s.v5.onnx --output yolov5s.v5.opt.onnx --in_tensor 167 --out_tensor 397,458,519 --verbose

---- Tengine YOLOv5 Optimize Tool ----

Input model      : yolov5s.v5.onnx
Output model     : yolov5s.v5.opt.onnx
Input tensor     : 167
Output tensor    : 397,458,519
[Quant Tools Info]: Step 0, load original onnx model from yolov5s.v5.onnx.
256
[Quant Tools Info]: Step 1, Remove the focus and postprocess nodes.
[Verbose] node_dict: {'130': 0, '135': 1, '140': 2, '145': 3, '150': 4, '155': 5, '160': 6, '165': 7, '166': 8, '167': 9, '168': 10, '169': 11, '170': 12, '171': 13, '172': 14, '173': 15, '174': 16, '175': 17, '176': 18, '177': 19, '178': 20, '179': 21, '180': 22, '181': 23, '182': 24, '183': 25, '184': 26, '185': 27, '186': 28, '187': 29, '188': 30, '189': 31, '190': 32, '191': 33, '192': 34, '193': 35, '194': 36, '195': 37, '196': 38, '197': 39, '198': 40, '199': 41, '200': 42, '201': 43, '202': 44, '203': 45, '204': 46, '205': 47, '206': 48, '207': 49, '208': 50, '209': 51, '210': 52, '211': 53, '212': 54, '213': 55, '214': 56, '215': 57, '216': 58, '217': 59, '218': 60, '219': 61, '220': 62, '221': 63, '222': 64, '223': 65, '224': 66, '225': 67, '226': 68, '227': 69, '228': 70, '229': 71, '230': 72, '231': 73, '232': 74, '233': 75, '234': 76, '235': 77, '236': 78, '237': 79, '238': 80, '239': 81, '240': 82, '241': 83, '242': 84, '243': 85, '244': 86, '245': 87, '246': 88, '247': 89, '248': 90, '249': 91, '250': 92, '251': 93, '252': 94, '253': 95, '254': 96, '255': 97, '256': 98, '257': 99, '258': 100, '259': 101, '260': 102, '261': 103, '262': 104, '263': 105, '264': 106, '265': 107, '266': 108, '267': 109, '268': 110, '269': 111, '270': 112, '271': 113, '272': 114, '273': 115, '274': 116, '275': 117, '276': 118, '277': 119, '278': 120, '279': 121, '280': 122, '281': 123, '282': 124, '283': 125, '284': 126, '285': 127, '286': 128, '287': 129, '288': 130, '289': 131, '294': 132, '295': 133, '296': 134, '297': 135, '298': 136, '299': 137, '300': 138, '301': 139, '302': 140, '303': 141, '304': 142, '305': 143, '306': 144, '307': 145, '308': 146, '309': 147, '310': 148, '311': 149, '312': 150, '313': 151, '314': 152, '319': 153, '320': 154, '321': 155, '322': 156, '323': 157, '324': 158, '325': 159, '326': 160, '327': 161, '328': 162, '329': 163, '330': 164, '331': 165, '332': 166, '333': 167, '334': 168, '335': 169, '336': 170, '337': 171, '338': 172, '339': 173, '340': 174, '341': 175, '342': 176, '343': 177, '344': 178, '345': 179, '346': 180, '347': 181, '348': 182, '349': 183, '350': 184, '351': 185, '352': 186, '353': 187, '354': 188, '355': 189, '356': 190, '357': 191, '358': 192, '359': 193, '360': 194, '361': 195, '362': 196, '363': 197, '364': 198, '365': 199, '366': 200, '367': 201, '368': 202, '369': 203, '370': 204, '371': 205, '372': 206, '373': 207, '374': 208, '375': 209, '376': 210, '377': 211, '400': 212, '401': 213, '402': 214, '407': 215, '409': 216, '411': 217, '413': 218, '415': 219, '420': 220, '422': 221, '424': 222, '426': 223, '431': 224, '432': 225, '442': 226, '443': 227, '466': 228, '467': 229, '468': 230, '473': 231, '475': 232, '477': 233, '479': 234, '481': 235, '486': 236, '488': 237, '490': 238, '492': 239, '497': 240, '498': 241, '508': 242, '509': 243, '532': 244, '533': 245, '534': 246, '539': 247, '541': 248, '543': 249, '545': 250, '547': 251, '552': 252, '554': 253, '556': 254, '558': 255, '563': 256, '564': 257, '574': 258, 'output': 259}
yolov5s-opt.py:80: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  output_pass = np.zeros((len(input_node)), dtype=np.int)
Traceback (most recent call last):
  File "yolov5s-opt.py", line 221, in <module>
    main()
  File "yolov5s-opt.py", line 184, in main
    new_nodes = cut_focus_output(old_node, in_tensor, out_tensor)
  File "yolov5s-opt.py", line 82, in cut_focus_output
    output_pass[node_dict[out_name[i]]] = 2
KeyError: '397'

qintian-319, 2023-02-06:
把onnx模型用https://netron.app/可视化，看这三个输出的代号，然后替换掉out_tensor的参数值
[image]
[image]
[image]