OAID/Tengine #1177: 源码编译（ACL）时的问题

Zhuxinpei, 2021-10-11:
在Ubuntu16.04下根据官方文档进行源码编译（ACL）时，在构建选项：
```
$ mkdir build-acl-arm64 && cd build-acl-arm64
$ cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/aarch64-linux-gnu.toolchain.cmake \
	-DTENGINE_ENABLE_ACL=ON ..
$ make -j4
$ make install
```
执行到make -j4时，出现如下错误：
```
[ 85%] Linking C shared library libtengine-lite.so
[ 85%] Built target tengine-lite
[ 85%] Building CXX object examples/CMakeFiles/tm_pipeline_enroll_face.dir/pipeline/test/test_face_enroll.cpp.o
[ 85%] Building CXX object examples/CMakeFiles/tm_crnn.dir/tm_crnn.cpp.o
[ 85%] Building CXX object benchmark/CMakeFiles/tm_benchmark.dir/common/timer.cc.o
[ 85%] Building CXX object examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/pipeline/test/test_pedestrian_distance_estimation.cpp.o
[ 85%] Building CXX object benchmark/CMakeFiles/tm_benchmark.dir/tm_benchmark.cc.o
In file included from /usr/local/Tengine/examples/pipeline/test/test_face_enroll.cpp:24:0:
/usr/local/Tengine/examples/pipeline/actor/image_stream.h:28:28: fatal error: opencv2/core.hpp: No such file or directory
compilation terminated.
examples/CMakeFiles/tm_pipeline_enroll_face.dir/build.make:75: recipe for target 'examples/CMakeFiles/tm_pipeline_enroll_face.dir/pipeline/test/test_face_enroll.cpp.o' failed
make[2]: *** [examples/CMakeFiles/tm_pipeline_enroll_face.dir/pipeline/test/test_face_enroll.cpp.o] Error 1
CMakeFiles/Makefile2:393: recipe for target 'examples/CMakeFiles/tm_pipeline_enroll_face.dir/all' failed
make[1]: *** [examples/CMakeFiles/tm_pipeline_enroll_face.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Building C object examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/common/tengine_operations.c.o
In file included from /usr/local/Tengine/examples/pipeline/test/test_pedestrian_distance_estimation.cpp:25:0:
/usr/local/Tengine/examples/pipeline/actor/pedestrian_detection.h:31:31: fatal error: opencv2/imgproc.hpp: No such file or directory
compilation terminated.
examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/build.make:75: recipe for target 'examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/pipeline/test/test_pedestrian_distance_estimation.cpp.o' failed
make[2]: *** [examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/pipeline/test/test_pedestrian_distance_estimation.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 85%] Building C object examples/CMakeFiles/tm_crnn.dir/common/tengine_operations.c.o
[ 85%] Linking CXX executable tm_benchmark
[ 85%] Built target tm_benchmark
CMakeFiles/Makefile2:445: recipe for target 'examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/all' failed
make[1]: *** [examples/CMakeFiles/tm_pipeline_estimate_ped_distance.dir/all] Error 2
[ 85%] Linking CXX executable tm_crnn
/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.9: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
examples/CMakeFiles/tm_crnn.dir/build.make:131: recipe for target 'examples/tm_crnn' failed
make[2]: *** [examples/tm_crnn] Error 1
CMakeFiles/Makefile2:419: recipe for target 'examples/CMakeFiles/tm_crnn.dir/all' failed
make[1]: *** [examples/CMakeFiles/tm_crnn.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```
推测大致错误包括：
```
fatal error: opencv2/core.hpp: No such file or directory
fatal error: opencv2/imgproc.hpp: No such file or directory
/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.9: error adding symbols: File in wrong format
```
可以确定我使用apt install libopencv-dev安装过opencv，且在/usr/include路径下有opencv和opencv2两个文件。在/usr/include/opencv2/core/中存在core.hpp文件以及在/usr/include/opencv2/imgproc/下存在imgproc.hpp文件。
我在编译方面的经验较少，无法找到更多的解决方法，麻烦给予更多指点！

lovehuanhuan, 2021-12-08:
cmake里面干掉examples

closed 2022-01-18 by Zhuxinpei

berylyellow, 2022-01-27:
@Zhuxinpei 你好，想问一下，你cmake 的时候，没有报错吗？我cmake都不行，
定位发现，source/CMakeLists.txt中有个TEngine_NEXT_LIB_SRCS 没有定义：
ADD_LIBRARY (${TENGINE_LITE_NAME}-static STATIC ${TEngine_NEXT_LIB_SRCS})
你知道怎么回事吗？

Zhuxinpei, 2022-01-28:
@berylyellow 您的问题我不太清楚，目前没有再继续使用Tengine。原来的编译问题通过不编译example的开关解决，即：
```
$ cmake -DCMAKE_TOOLCHAIN_FILE=../toolchains/aarch64-linux-gnu.toolchain.cmake \
	-DTENGINE_ENABLE_ACL=ON -DTENGINE_BUILD_BENCHMARK=OFF -DTENGINE_BUILD_EXAMPLES=OFF ..
```
建议您可以重提issue请官方来解答。

liujiaxing7, 2022-04-22:
> @Zhuxinpei 你好，想问一下，你cmake 的时候，没有报错吗？我cmake都不行， 定位发现，source/CMakeLists.txt中有个TEngine_NEXT_LIB_SRCS 没有定义： ADD_LIBRARY (${TENGINE_LITE_NAME}-static STATIC ${TEngine_NEXT_LIB_SRCS}) 你知道怎么回事吗？

应该是cmake版本不对