OAID/Tengine #1208: Compiled the libtengine-lite.so successfully with C308X SDK6.4.4.3, but error occured when running on C308X.

48313758a, 2021-11-12:
I have tried compile with TIM-VX  and sdk  reference to RV1109/RV1126.
My galcore.ko inquired by dmesg | grep Galcore is [   69.264789@1] Galcore version 6.4.4.3.310723.
I also update the following files:
libGAL.so          libOpenCL.so   libOvx12VXCBinary.so
libarchmodelSw.so  libNNArchPerf.so   libOpenVXC.so  libovxlib.so
libArchModelSw.so  libNNGPUBinary.so  libOpenVX.so   libVivanteOpenCL.so
libCLC.so          libNNVXCBinary.so  libOpenVXU.so  libVSC.so

But when I run tm_yolov5s_i8 that I made:
./tm_yolov5s_i8 -m ./yolov5s_uint8.tmfile -i ./dog.jpg -r 5

ERROR happend:
tengine-lite library version: 1.5-dev

Tengine Fatal: Pre-run subgraph(0) on TIMVX failed.
Tengine: Scheduler(sync) prerun failed.
Prerun multithread graph failed.

ChungTak, 2021-11-26:
What is tm_yolov5s_i8?  You should use tm_yolov5s_timvx

48313758a, 2021-12-06:
> 

I used tm_yolov5s_timvx.cpp, and the model is yolov5s_uint8, my target .os named tm_yolov5s_i8.

48313758a, 2021-12-06:
@ChungTak I used tm_yolov5s_timvx.cpp, and the model is yolov5s_uint8, my target .os named tm_yolov5s_i8.@ChungTak