OAID/Tengine #1209: (English/中文)How to fix the error "Segmentation fault" when execute the example command "tm_classification_acl" in the Orange Pi 4(RK3399)?

zhangweiwahaha042, 2021-11-15:
■description
I'm so sorry I need you help.
the error "Segmentation fault" occur when I execute the example command "tm_classification_acl".
Could you kindly please tell me how to fix that error or give me a good debug suggestion for that error?

■system information
・hardware platform
Orange Pi 4(RK3399)
・system
ubuntu 18.04
・uname -a
orangepi@orangepi4:~$ uname -a
Linux orangepi4 4.4.179-rk3399 #4 SMP Tue Nov 9 11:24:14 JST 2021 aarch64 aarch64 aarch64 GNU/Linux
・ARM Compute Library(ACL)
version:v20.05
・Tengine Lite
the last commit id : 2b5b451a07e9016a6dd516a47eb4bf60f007eb4d
Date: Mon Nov 15 14:13:45 2021 +0800

■procedure
(1). download and build the source code of the ACL(Arm Compute Library) and Tengine Lite refer to the following site.
https://tengine-docs.readthedocs.io/en/latest/source_compile/compile_acl.html
(2). download the "mobilenet.tmfile" and the "ssd_dog.jpg" from the following site.
https://pan.baidu.com/s/1JsitkY6FVV87Kao6h5yAmg
(key:7ke5)
(3). the error "Segmentation fault" occur when execute the command "tm_classification_acl"
orangepi@orangepi4:~/Tengine-Lite$ ./build-acl-arm64/examples/tm_classification_acl -m mobilenet.tmfile -i ssd_dog.jpg -t 1 -r 10
Image height not specified, use default 227
Image width not specified, use default  227
Scale value not specified, use default  1.0, 1.0, 1.0
Mean value not specified, use default   104.0, 116.7, 122.7
tengine-lite library version: 1.5-dev
ACL initialized
Segmentation fault

■Additional information
there is no error occur when execute the command "tm_classification"
orangepi@orangepi4:~/test_tengine_lite/tengine-lite/Tengine-Lite$ ./build-acl-arm64/examples/tm_classification -m mobilenet.tmfile -i ssd_dog.jpg -t 1 -r 10
Image height not specified, use default 224
Image width not specified, use default  224
Scale value not specified, use default  0.0, 0.0, 0.0
Mean value not specified, use default   104.0, 116.7, 122.7
tengine-lite library version: 1.5-dev
model file : mobilenet.tmfile
image file : ssd_dog.jpg
img_h, img_w, scale[3], mean[3] : 224 224 , 0.017 0.017 0.017, 104.0 116.7 122.7
Repeat 10 times, thread 1, avg time 136.40 ms, max_time 192.38 ms, min_time 118.96 ms
\--------------------------------------
11.015098, 249
9.838374, 248
9.687143, 250
9.655354, 196
9.295464, 198
\--------------------------------------

zhangweiwahaha042, 2021-11-15:
=========================================== 中文描述 ===========================================
■问题描述
很抱歉我这边遇到了一个问题需要你们的帮助。
当我试着运行实例命令"tm_classification_acl"的时候，发生了致命错误"Segmentation fault"。
您能告诉我，怎样解决这个错误吗？或者告诉我一些调试的方法，我自己也可以尝试进行解决。

■系统情报
・硬件板子
Orange Pi 4(RK3399)
・操作系统
ubuntu 18.04
・uname -a
orangepi@orangepi4:~$ uname -a
Linux orangepi4 4.4.179-rk3399 #4 SMP Tue Nov 9 11:24:14 JST 2021 aarch64 aarch64 aarch64 GNU/Linux
・ARM Compute Library(ACL)
version:v20.05
・Tengine Lite
the last commit id : 2b5b451a07e9016a6dd516a47eb4bf60f007eb4d
Date: Mon Nov 15 14:13:45 2021 +0800

■执行步骤
(1). 参考下面的网站信息，下载并编译ACL(Arm Compute Library)和Tengine Lite的代码
https://tengine-docs.readthedocs.io/en/latest/source_compile/compile_acl.html
(2). 从下面的网站上下载并获取"mobilenet.tmfile"和"ssd_dog.jpg"
https://pan.baidu.com/s/1JsitkY6FVV87Kao6h5yAmg
(提取码:7ke5)
(3). 当我运行实列程序"tm_classification_acl"的时候，发生了致命错误"Segmentation fault"
orangepi@orangepi4:~/Tengine-Lite$ ./build-acl-arm64/examples/tm_classification_acl -m mobilenet.tmfile -i ssd_dog.jpg -t 1 -r 10
Image height not specified, use default 227
Image width not specified, use default  227
Scale value not specified, use default  1.0, 1.0, 1.0
Mean value not specified, use default   104.0, 116.7, 122.7
tengine-lite library version: 1.5-dev
ACL initialized
Segmentation fault

■补充说明
当我运行实列程序"tm_classification"的时候，执行没有问题。
orangepi@orangepi4:~/test_tengine_lite/tengine-lite/Tengine-Lite$ ./build-acl-arm64/examples/tm_classification -m mobilenet.tmfile -i ssd_dog.jpg -t 1 -r 10
Image height not specified, use default 224
Image width not specified, use default  224
Scale value not specified, use default  0.0, 0.0, 0.0
Mean value not specified, use default   104.0, 116.7, 122.7
tengine-lite library version: 1.5-dev
model file : mobilenet.tmfile
image file : ssd_dog.jpg
img_h, img_w, scale[3], mean[3] : 224 224 , 0.017 0.017 0.017, 104.0 116.7 122.7
Repeat 10 times, thread 1, avg time 136.40 ms, max_time 192.38 ms, min_time 118.96 ms
\--------------------------------------
11.015098, 249
9.838374, 248
9.687143, 250
9.655354, 196
9.295464, 198
\--------------------------------------

zhangweiwahaha042, 2021-11-19:
Dear Tengine:
I clean all the environment, re-build the source and do the test again, the test result is OK, so I close this ticket.
Thank you all the same.

closed 2021-11-19 by zhangweiwahaha042

Kellin233, 2022-07-14:
> Dear Tengine: I clean all the environment, re-build the source and do the test again, the test result is OK, so I close this ticket. Thank you all the same.

I got the same problem now.  And I did what you did but still have the problem. Can you please tell me more details about how you solve the problem or give me some suggestions on solving the problem?