OAID/Tengine #1213: 请问在OpenCV+Tengine里如何使能"GPU FP16/FP32模式"?

zhangweiwahaha042, 2021-11-17:
■问题描述
很抱歉，打扰了，我现在需要你们的帮助。
现在板子可以在CPU上运行OpenCV+Tengine(编译OpenCV时，加上cmake参数"-DWITH_TENGINE=ON")，如果想进一步，使用GPU进行加速，应该怎样操作呢？
<详细说明>
参考分支"tengine-lite"代码里的"acl.md"的说明，好像可以通过设置create_context()，add_context_device()等API，使能"GPU FP16/FP32模式"以实现GPU的工作。
但是，现在OpenCV的代码里自动下载的是分支"tengine-opencv"的代码，且该代码里并没有上述这种可以"使能GPU FP16/FP32模式"的API。
请问，"tengine-opencv"的情况，可以把GPU用起来吗？
如果可以，应该怎样设置呢？
如果不可以，我将编译好的"tengine-lite"的库和头文件合入到opencv代码后，再参考"acl.md"进行GPU的工作，这个方案可行吗？

■系统情报
・硬件平台
Orange Pi 4
・CPU
Rockchip RK3399:双核Cortex-A72+四核Cortex-A53
・GPU
Mali-T864
・操作系统
ubuntu 18.04
・uname -a
Linux orangepi4 4.4.179-rk3399 #4 SMP Tue Nov 9 11:24:14 JST 2021 aarch64 aarch64 aarch64 GNU/Linux
・OpenCV版本
opencv_4.5.4
最后一个提交commit id:4223495e6cd67011f86b8ecd9be1fa105018f3b1
最后一个提交时间:2021年10月09日

RudyFoo, 2021-11-17:
回答说：不行。
如果你想尝试，OpenCV调用tengine-lite的函数接口，从顶层到中间层要改很多接口。结果要等人尝鲜才知道。

zhangweiwahaha042, 2021-11-17:
@RudyFoo 谢谢你的回复。

closed 2021-11-17 by zhangweiwahaha042