OAID/Tengine #1220: 适配yolov5版本6模型问题

ChungTak, 2021-11-26:
一、问题描述
适配yolov5版本6模型，推理出现错误`Error: input elem num(1612800) != reshaped elem num(1612790)`

二、操作步骤：
1. 导出onnx： `python3 export.py --weights weights/yolov5n.pt --img 640 --batch 1  --opset 10  --simplify`
2. 使用netron 查看yolov5n.onnx的结构,新的结构已经没有focus所以要用最新脚本添加--cut_focus参数， out_tensor是 337,390,443，
in_tensor 以前focus后的167不存在了，填images 
```
yolov5s-opt.py   --input yolov5n.onnx   --output yolov5n.v6.opt.onnx   --in_tensor images   --out_tensor 337,390,443   --cut_focus
```
3. 转换为tmfile。下载release v1.5版本下的convert_tool-x86_64.AppImage文件
convert_tool-x86_64.AppImage -f onnx -m yolov5n.v6.opt.onnx -o yolov5n.v6.tmfile
4.推理,确认拉取最新的代码 #define YOLOV5_VERSION6 的值 1 表示 版本6,0 表示版本5
```
tm_yolov5s -m yolov5n.v6.tmfile -i ssd_dog.jpg 
```
出现错误

```
Error: input elem num(1612800) != reshaped elem num(1612790)
Tengine FATAL: Infer node(id: 296, op: Reshape) shape failed.
Tengine: Infer shape of graph failed(-1).
Prerun multithread graph failed.
```
三、运行环境
1.操作系统：Ubuntu20.04
2.python3.8 
3.pytourch 1.7.1
4.Tengine-Lite 最新代码 commit: b63495e217913d251d6e7ec0df9ecabf914718c8

deepage, 2021-11-26:
看了您的tmfile之后，发现您并没有去掉后处理，建议在onnx导出的步骤中加入--train参数用来去除模型的后处理操作。

ChungTak, 2021-11-26:
感谢超神！ 导出添加--train 后重新转换就可以了！
导出后结构有改变，最终完美运行的参数是
```
python3 yolov5s-opt.py   --input yolov5n.onnx   --output yolov5n.v6.opt.onnx   --in_tensor images   --out_tensor output,351,365   --cut_focus
```

ChungTak, 2021-11-26:
量化操作与之前版本有区别，-g 参数 3,640,640  -k参数不需要
```
./quant_tool_uint8 -m  yolov5n.v6.tmfile -i  /datavol/test-images/temp -o  yolov5n.uint8.tmfile -g 3,640,640 -w 0,0,0 -s 0.003921,0.003921,0.003921 -c 0  -y 640,640
```

[量化推理代码tm_yolov5n_uint8.cpp.zip](https://github.com/OAID/Tengine/files/7607993/tm_yolov5n_uint8.cpp.zip)

closed 2021-11-26 by ChungTak

lumiaxth, 2022-08-24:
请问量化操作中，参数-w -s的值是依照什么来设定的？