OAID/Tengine #1221: retinaface模型如何量化?

ChungTak, 2021-11-29:
一、问题描述
官方 retinaface.tmfile 模型量化uin8后，推理检测不出人脸。
查看模型看输入是type: float32[] ，没有指定大小，这种模型量化的参数应该怎样填？
二、操作步骤：
不知道参数怎样填，尝试过以下参数
`./quant_tool_uint8 -m retinaface.tmfile  -i Face_Datasets -o retinaface.uint8.tmfile -g 3,640,640 `
或者
`./quant_tool_uint8 -m retinaface.tmfile  -i Face_Datasets -o retinaface.uint8.tmfile  -g 3,416,416 -w 104.007,116.669,122.679 -s 0.017,0.017,0.017`
然后使用官方的推理例子
`tm_retinaface_uint8  -m ../test_models/uin8/retinaface.uint8.tmfile -i face5.jpg`
返回结果
```
img_h, img_w : 240, 320
Repeat 1 times, thread 1, avg time 64.90 ms, max_time 64.90 ms, min_time 64.90 ms
--------------------------------------
detected face num: 0
```

三、运行环境
1.操作系统：Ubuntu20.04
2.python3.8
3.pytourch 1.7.1
4.Tengine-Lite 最新代码 commit: b63495e

lovehuanhuan, 2021-12-06:
./quant_tool_uint8 -m retinaface.tmfile -i Face_Datasets -o retinaface.uint8.tmfile -g 3,416,416 -w 0,0,0-s 1,1,1

ChungTak, 2021-12-06:
@lovehuanhuan 哇～真的可以了！非常感谢～～请教一下大佬～～
模型量化时的 -w mean 和 -s  scale 是怎样确定这个值的？
我在retinaface源代码和tm_retinaface_timvx 推理代码里没找到0,0,0和1,1,1这些值～

closed 2021-12-08 by ChungTak