OAID/Tengine #1235: yolov5s on TensorRT failed

wfeii1980, 2021-12-20:
# env

[image]

# error

```
$ tm_yolov5s -m yolov5s.tmfile -i images/ssd_dog.jpg
tengine-lite library version: 1.5-dev
Tengine Error: 4: [network.cpp::validate::2437] Error Code 4: Internal Error (Repeated tensor name: 203 (tensors must have distinct names))
Tengine: Can not Build engine, please check config.
Tengine Fatal: Pre-run subgraph(20) on TensorRT failed.
Tengine: Scheduler(sync) prerun failed.
Prerun multithread graph failed.
```

wfeii1980, 2021-12-21:
```shell
$ /usr/src/tensorrt/bin/trtexec --onnx=model_opt.onnx
&&&& RUNNING TensorRT.trtexec [TensorRT v8001] # /usr/src/tensorrt/bin/trtexec --onnx=model_opt.onnx
[12/21/2021-09:47:50] [I] === Model Options ===
[12/21/2021-09:47:50] [I] Format: ONNX
[12/21/2021-09:47:50] [I] Model: model_opt.onnx
[12/21/2021-09:47:50] [I] Output:
[12/21/2021-09:47:50] [I] === Build Options ===
[12/21/2021-09:47:50] [I] Max batch: explicit
[12/21/2021-09:47:50] [I] Workspace: 16 MiB
[12/21/2021-09:47:50] [I] minTiming: 1
[12/21/2021-09:47:50] [I] avgTiming: 8
[12/21/2021-09:47:50] [I] Precision: FP32
[12/21/2021-09:47:50] [I] Calibration: 
[12/21/2021-09:47:50] [I] Refit: Disabled
[12/21/2021-09:47:50] [I] Sparsity: Disabled
[12/21/2021-09:47:50] [I] Safe mode: Disabled
[12/21/2021-09:47:50] [I] Restricted mode: Disabled
[12/21/2021-09:47:50] [I] Save engine: 
[12/21/2021-09:47:50] [I] Load engine: 
[12/21/2021-09:47:50] [I] NVTX verbosity: 0
[12/21/2021-09:47:50] [I] Tactic sources: Using default tactic sources
[12/21/2021-09:47:50] [I] timingCacheMode: local
[12/21/2021-09:47:50] [I] timingCacheFile: 
[12/21/2021-09:47:50] [I] Input(s)s format: fp32:CHW
[12/21/2021-09:47:50] [I] Output(s)s format: fp32:CHW
[12/21/2021-09:47:50] [I] Input build shapes: model
[12/21/2021-09:47:50] [I] Input calibration shapes: model
[12/21/2021-09:47:50] [I] === System Options ===
[12/21/2021-09:47:50] [I] Device: 0
[12/21/2021-09:47:50] [I] DLACore: 
[12/21/2021-09:47:50] [I] Plugins:
[12/21/2021-09:47:50] [I] === Inference Options ===
[12/21/2021-09:47:50] [I] Batch: Explicit
[12/21/2021-09:47:50] [I] Input inference shapes: model
[12/21/2021-09:47:50] [I] Iterations: 10
[12/21/2021-09:47:50] [I] Duration: 3s (+ 200ms warm up)
[12/21/2021-09:47:50] [I] Sleep time: 0ms
[12/21/2021-09:47:50] [I] Streams: 1
[12/21/2021-09:47:50] [I] ExposeDMA: Disabled
[12/21/2021-09:47:50] [I] Data transfers: Enabled
[12/21/2021-09:47:50] [I] Spin-wait: Disabled
[12/21/2021-09:47:50] [I] Multithreading: Disabled
[12/21/2021-09:47:50] [I] CUDA Graph: Disabled
[12/21/2021-09:47:50] [I] Separate profiling: Disabled
[12/21/2021-09:47:50] [I] Time Deserialize: Disabled
[12/21/2021-09:47:50] [I] Time Refit: Disabled
[12/21/2021-09:47:50] [I] Skip inference: Disabled
[12/21/2021-09:47:50] [I] Inputs:
[12/21/2021-09:47:50] [I] === Reporting Options ===
[12/21/2021-09:47:50] [I] Verbose: Disabled
[12/21/2021-09:47:50] [I] Averages: 10 inferences
[12/21/2021-09:47:50] [I] Percentile: 99
[12/21/2021-09:47:50] [I] Dump refittable layers:Disabled
[12/21/2021-09:47:50] [I] Dump output: Disabled
[12/21/2021-09:47:50] [I] Profile: Disabled
[12/21/2021-09:47:50] [I] Export timing to JSON file: 
[12/21/2021-09:47:50] [I] Export output to JSON file: 
[12/21/2021-09:47:50] [I] Export profile to JSON file: 
[12/21/2021-09:47:50] [I] 
[12/21/2021-09:47:50] [I] === Device Information ===
[12/21/2021-09:47:50] [I] Selected Device: Xavier
[12/21/2021-09:47:50] [I] Compute Capability: 7.2
[12/21/2021-09:47:50] [I] SMs: 6
[12/21/2021-09:47:50] [I] Compute Clock Rate: 1.109 GHz
[12/21/2021-09:47:50] [I] Device Global Memory: 7773 MiB
[12/21/2021-09:47:50] [I] Shared Memory per SM: 96 KiB
[12/21/2021-09:47:50] [I] Memory Bus Width: 256 bits (ECC disabled)
[12/21/2021-09:47:50] [I] Memory Clock Rate: 1.109 GHz
[12/21/2021-09:47:50] [I] 
[12/21/2021-09:47:50] [I] TensorRT version: 8001
[12/21/2021-09:47:52] [I] [TRT] [MemUsageChange] Init CUDA: CPU +353, GPU +0, now: CPU 371, GPU 4584 (MiB)
[12/21/2021-09:47:52] [I] Start parsing network model
[12/21/2021-09:47:52] [I] [TRT] ----------------------------------------------------------------
[12/21/2021-09:47:52] [I] [TRT] Input filename:   model_opt.onnx
[12/21/2021-09:47:52] [I] [TRT] ONNX IR version:  0.0.6
[12/21/2021-09:47:52] [I] [TRT] Opset version:    12
[12/21/2021-09:47:52] [I] [TRT] Producer name:    pytorch
[12/21/2021-09:47:52] [I] [TRT] Producer version: 1.8
[12/21/2021-09:47:52] [I] [TRT] Domain:           
[12/21/2021-09:47:52] [I] [TRT] Model version:    0
[12/21/2021-09:47:52] [I] [TRT] Doc string:       
[12/21/2021-09:47:52] [I] [TRT] ----------------------------------------------------------------
[12/21/2021-09:47:53] [W] [TRT] onnx2trt_utils.cpp:364: Your ONNX model has been generated with INT64 weights, while TensorRT does not natively support INT64. Attempting to cast down to INT32.
[12/21/2021-09:47:53] [W] [TRT] onnx2trt_utils.cpp:390: One or more weights outside the range of INT32 was clamped
[12/21/2021-09:47:53] [I] [TRT] No importer registered for op: HardSwish. Attempting to import as plugin.
[12/21/2021-09:47:53] [I] [TRT] Searching for plugin: HardSwish, plugin_version: 1, plugin_namespace: 
[12/21/2021-09:47:53] [E] [TRT] 3: getPluginCreator could not find plugin: HardSwish version: 1
[12/21/2021-09:47:53] [E] [TRT] ModelImporter.cpp:720: While parsing node number 1 [HardSwish -> "407"]:
[12/21/2021-09:47:53] [E] [TRT] ModelImporter.cpp:721: --- Begin node ---
[12/21/2021-09:47:53] [E] [TRT] ModelImporter.cpp:722: input: "733"
output: "407"
name: "Sigmoid_42"
op_type: "HardSwish"

[12/21/2021-09:47:53] [E] [TRT] ModelImporter.cpp:723: --- End node ---
[12/21/2021-09:47:53] [E] [TRT] ModelImporter.cpp:726: ERROR: builtin_op_importers.cpp:4643 In function importFallbackPluginImporter:
[8] Assertion failed: creator && "Plugin not found, are the plugin name, version, and namespace correct?"
[12/21/2021-09:47:53] [E] Failed to parse onnx file
[12/21/2021-09:47:53] [I] Finish parsing network model
[12/21/2021-09:47:53] [E] Parsing model failed
[12/21/2021-09:47:53] [E] Engine creation failed
[12/21/2021-09:47:53] [E] Engine set up failed
&&&& FAILED TensorRT.trtexec [TensorRT v8001] # /usr/src/tensorrt/bin/trtexec --onnx=model_opt.onnx
```

wfeii1980, 2021-12-21:
[image]
[image]

wfeii1980, 2021-12-21:
```
diff --git a/source/device/tensorrt/trt_limit.hpp b/source/device/tensorrt/trt_limit.hpp
index 0a9402e7..396d2834 100644
--- a/source/device/tensorrt/trt_limit.hpp
+++ b/source/device/tensorrt/trt_limit.hpp
@@ -88,7 +88,7 @@ const int trt_supported_ops[] = {
 //        OP_GRU,
 #endif
     //        OP_HARDSIGMOID,
-    //        OP_HARDSWISH,                   // Not supported, last checked version 7.1.3
+    OP_HARDSWISH,                   // Not supported, last checked version 7.1.3
     OP_INPUT,
     OP_INSTANCENORM,
     OP_INTERP, // should be as UpSample
```

ringwraith, 2022-09-16:
> ```
> diff --git a/source/device/tensorrt/trt_limit.hpp b/source/device/tensorrt/trt_limit.hpp
> index 0a9402e7..396d2834 100644
> --- a/source/device/tensorrt/trt_limit.hpp
> +++ [...]

Thanks for the solution!