OAID/Tengine #1238: Convert to model but  Loss some  op

qqqtankqqq, 2021-12-22:
I  convert my onnx model to tmfile  by using Tengine-convert-tool  successfully. 
But tmfile loss some operation like relu and BatchNormalization(1d).

[image]
[image]

I already try to  create just one BatchNormalization model. it works.

does anyone have similar problem?

xiguadong, 2021-12-23:
convert-tool has optimized the model to decrease the caculate(the other forward frameworks also has this step). You could google bn-fold.

BUG1989, 2022-01-05:
has been merged into convolution

closed 2022-01-05 by BUG1989